proto17/dji_droneid #4: gr-drone-update block error

alphafox02, 2022-05-17:
With GNU Radio 3.8.1 the following shows on loading of the drone_id_test.grc file, but not sure if it's a negative impact as the flow graph still loads.

`WARNING:gnuradio.grc.core.platform:droneid_misc_utils.block.yml             block: error: Value type 'NoneType' for 'parameters' not in valid types
WARNING:gnuradio.grc.core.platform:droneid_misc_utils.block.yml             block: error: Value type 'NoneType' for 'inputs' not in valid types
WARNING:gnuradio.grc.core.platform:droneid_misc_utils.block.yml             block: error: Value type 'NoneType' for 'outputs' not in valid types
INFO:gnuradio.grc.core.platform:YAML schema check failed for: /usr/local/share/gnuradio/grc/blocks/droneid_misc_utils.block.yml
`
Does this replace the use of the turbo encoder/octave script? Just reading over all the new commits, can't wait to try again.

proto17, 2022-05-17:
Hmm, I wonder if it's best to just remove that file.  It's a GRC file for a C++ class that isn't actually a block you can use in the flow graph.  You can try deleting the .yml file for that block and seeing if GNU Radio is happy again.  I assumed I could just leave the input and output sections empty and be okay.  

The gr-droneid-update branch does have the turbo decoder in GNU Radio [1] and should output hex when it can successfully demod a frame.  Trouble is that the demod really sucks.  I didn't do anything to fix the CFO estimate nor did I add anything for equalization.  This was just a quick and *dirty* way to get to bytes.  Even in really good SNR you will see constant demod failures.  I hope to have something ready for starting time offset (STO) and integer frequency offset (IFO) in the coming weeks.

[1] https://github.com/proto17/dji_droneid/blob/gr-droneid-update/gnuradio/gr-droneid/lib/demodulation_impl.cc#L418

alphafox02, 2022-05-17:
Deleting the yml file made GNU Radio happy again. 

I ran my IQ file back through the flow graph and can see that it works, I'm getting output in the bottom left corner. I don't know what all I can copy/paste here, but this is basically what I'm seeing minus the hex output. The output, and I've not looked super close to compare, appears different then when running the IQ file through the octave script. Great to hear it's moving towards an all in one setup. 

`Executing: /usr/bin/python3 -u /home/dragon/Desktop/drone_id_demod_test.py

FFT SIZE: 2048, sample rate: 3.072e+07
Found burst @ 100237942 / 100237941
Sending message with 23856 elements (23856) starting at 100237941
Max: 6736 offset: 6736 - 0
short: 144 long: 160 fft: 2048
((start_idx . 0))
ZC start index: 6736, est start index: 0
.................CRC Check Failed!
..................
CRC Check Failed!
Time: 0.000955394
Found burst @ 119898722 / 119898721
Sending message with 23856 elements (23856) starting at 119898721
Max: 6737 offset: 6736 - 1
short: 144 long: 160 fft: 2048
((start_idx . 1))
ZC start index: 6737, est start index: 1
.............................
CRC Check Failed!
Time: 0.000769564
........................................................................
>>> Done`

alphafox02, 2022-05-17:
I should've mentioned, or i may have in another ticket, the bursts that had appeared before in my /tmp directory would never reveal anything when ran through the octave script. However, running the entire IQ file through would get a good hex with correct information.

closed 2022-05-17 by alphafox02

proto17, 2022-05-17:
Glad to hear it's not just immediately crashing :smile: I'm not all that surprised about the CRC failures.  You will see the hex output even if the frame fails, so don't assume that the hex is valid if you see the CRC failure messages.  The CRC here is the 24 bit CRC that goes with the Turbo decoder, not the 16 bit CRC for the DroneID frame.