proto17/dji_droneid #9: DJI DroneID frames & burst generation and demodulation test

Retinaaa, 2022-06-15:
Hi there,
Very interesting work and write-up. I'd love to test 
[proto17](https://github.com/proto17) /
[dji_droneid](https://github.com/proto17/dji_droneid)
First, I generated the frames with the appropriate hypothetical data and generated the DJI DroneID burst.
`clc
clear 
close
sample_rate=30.72e6;

temp_path1="/home/retina/Droneid/dji_droneid-main/matlab/updated_scripts/transmit";
add_turbo_path1="/home/retina/Droneid/dji_droneid-main/matlab/updated_scripts/transmit/add_turbo";

varargin2 = {'SequenceNumber', 7.0, 'StateInfo0', uint8(9),'StateInfo1', uint8(1),'SerialNumber','264CG2TZ0A2FL5',...
    'Longitude', 45.0,'Latitude', 65.0,'Height' , int16(100),'Altitude', int16(55),...
    'VelocityNorth', int16(20),'VelocityEast', int16(40.0),'VelocityUp', int16(60.0),'Yaw', int16(80.0),...
    'PhoneAppGPSTime', uint64(900),'PhoneAppLatitude', 12.0,'HomeLongitude', 22.0,...
    'ProductType', uint8(1), 'UUID','UUID must be 19 cha'}

% function [samples] = create_burst(sample_rate, frame_configuration, add_turbo_path, temp_path, show_debug_plots)
Tsampls=create_burst(sample_rate,varargin2,add_turbo_path1,temp_path1,1)

%% save file===============================================================    
file_name = 'landing.dat';
vec_for_wrt = single(Tsampls);
Tsampls_bin = fullfile(temp_path1, 'TsamplsF.dat');              % Will write Tsampls bytes here
A = nan(2,length(vec_for_wrt));
for cnt = 1 : length(vec_for_wrt)
    A(1,cnt) = real(vec_for_wrt(cnt));
    A(2,cnt) = imag(vec_for_wrt(cnt));
end
B = reshape(A,[],1);
file_id = fopen(Tsampls_bin,'w');
assert(file_id ~= -1, 'Could not open output temp file "%s"', Tsampls_bin);
fwrite(file_id,B,'float');
fclose(file_id);`
[image]
[image]

Then I gave a signal to the file `process_file.m`
But even though the correlation peak output is greater than 0.7, the following error was reported
I either get:
[image]

Is the issue with running the bursts through MATLAB is that some amount of padding either side of the burst is required in order to do some of the operations. , or have I configured something incorrectly?

Thanks for all your work, very interesting repo!

proto17, 2022-06-15:
Glad to see you're getting good samples out!  The issue you're seeing is that the `process_file.m` script wants a little bit of padding to the left and right of the burst.  The `extract_bursts_from_file` function is provided a number of samples worth of padding that it should extract which is applied to the beginning and end of the burst [1].  [2] is where I set the padding to be the filter length so that I have enough samples at both ends to prevent the low pass filter from messing up the important samples.  So, if you want to run the burst back through `process_file.m` you need to add some noise or just zeros to the beginning and end of the burst.

[1] https://github.com/proto17/dji_droneid/blob/main/matlab/updated_scripts/extract_bursts_from_file.m#L23
[2] https://github.com/proto17/dji_droneid/blob/main/matlab/updated_scripts/process_file.m#L51

Retinaaa, 2022-06-15:
The padding worked, Thanks once again for your great work!

closed 2022-06-21 by Retinaaa