proto17/dji_droneid #12: Cross-correlation of  zc sequence and samples looks error

jianjianliu0918, 2022-08-16:
Hello everyone
I tried run the uav iq data in code matlab/updated_scripts/process_file.m.
The cross-correlation of  zc sequence and samples looks wrong. The results dosen't have a good correlation peak.
The relevant parameters are as follows:
Data record board: usrp b210, sample rate 30.72MHz
UAV type: DJI AIR2, transmission system is DJI OCUSYNC 2.0.

edit the matlab code as follows:
file_sample_rate = 30.72e6;
chunk_size = 10e6;

Waterfalls:
[image]

Spetrum:
[image]

Result of cross correlation:
[image]

proto17, 2022-08-16:
The number of frames you are seeing makes me believe that you are looking at the downlink signal and not the DroneID signal.  The DroneID signal is not nearly that active.  I don't recall exactly, but I think it's something like every 600 milliseconds.  The downlink signal can have the same bandwidth, but does not have the same structure, and likely doesn't use the same ZC sequences.  Initially I thought you might be seeing the uplink, but I'm pretty sure that the uplink is ~ 1.5 MHz wide, where the downlink is selectable to something like 10 or 20 MHz.

jianjianliu0918, 2022-08-16:
> The number of frames you are seeing makes me believe that you are looking at the downlink signal and not the DroneID signal. The DroneID signal is not nearly that active. I don't recall exactly, but [...]

Yes, I analyzed the downlink drones video transmitter signal above, it seems that I have misunderstood, Tomorrow I will re-analyze the uplink signal.
Thanks!

proto17, 2022-08-16:
The uplink will not work with this code either.  This code only supports DroneID which comes from the drone, not the controller.  There is a list [1] of frequencies that are known to carry DroneID signals for reference.  The bursts that you see should be very infrequent, only stay on the same frequency for ~ 20-30 bursts, and hop around quite a bit.

[1] https://github.com/proto17/dji_droneid#frequencies

jianjianliu0918, 2022-08-17:
> The uplink will not work with this code either. This code only supports DroneID which comes from the drone, not the controller. There is a list [1] of frequencies that are known to carry DroneID [...]

I have record data as a frequency list, and found a strong corrleation with zc seq.

DJI Mavic 3,  central frequnce is 2445.5MHz:
cross correlation results (I don't know why this result are not normalized, so I set the thresold to 14):
[image]

corresponding position：
[image]

Then I calculate the correlation between the extracted CP and OFDM frame structure data，and the result doesn't seem to be correct. Normally, this results should have multiple peaks associated with the CP position.
[image]

Unsurprisingly,  the constellation results show that the signal demodulation process has errors.
[image]

I don't know if there is a difference between  ocusync2.0 and ocusync3.0(Mavic 3). Or something went wrong in signal processing.

jianjianliu0918, 2022-08-17:
An interesting phenomenon, I intercepted this signal, and then used the ofdm blind detection method to detect the CP position. 
Results show as follows:
[image]
Similar to ocusync 2.0 , two OFDM signals are also placed in front of the ZC sequence.
The entire signal has 8 cyclic prefixes (including the ZC seq).
These results make me suspect that this signal is the drone id signal for ocusync 3.0.

proto17, 2022-08-17:
The MATLAB code in the `main` branch does use a normalized cross correlation to find the ZC sequence.  If you're using the MATLAB code from the GNU Radio branches then that might explain what you're seeing.  

The DroneID signal will either contain 9 or 8 OFDM symbols.  I have a Mini 2 that sends 9 symbols, but most of what I have seen from other collects is 8.  The code in this repo assumes that all 9 are there, but only actually does anything with the last 8.  

I don't currently do blind estimation using the cyclic prefix just because I have the ZC sequence which gives me a pretty accurate starting point as opposed to the very rough start index of the cyclic prefixes.  I do (in one version of the code somewhere) use the cyclic prefixes for coarse frequency offset estimation though.

My advice is to try the latest `main` branch of the repo and see if that helps with the correlation and ultimate data carrier extraction.

jianjianliu0918, 2022-08-19:
> 

> The MATLAB code in the `main` branch does use a normalized cross correlation to find the ZC sequence. If you're using the MATLAB code from the GNU Radio branches then that might explain what you're seeing.
> 
> The DroneID signal will either contain 9 or 8 OFDM symbols. I have a Mini 2 that sends 9 symbols, but most of what I have seen from other collects is 8. The code in this repo assumes that all 9 are there, but only actually does anything with the last 8.
> 
> I don't currently do blind estimation using the cyclic prefix just because I have the ZC sequence which gives me a pretty accurate starting point as opposed to the very rough start index of the cyclic prefixes. I do (in one version of the code somewhere) use the cyclic prefixes for coarse frequency offset estimation though.
> 
> My advice is to try the latest `main` branch of the repo and see if that helps with the correlation and ultimate data carrier extraction.
> 
> EDIT:
> 
> For reference, see [1] which is where the normalized cross correlation function is called. In previous versions the correlation was done with the `filt` function. This did not produce normalized results, so it was changed to `xcorr(x, y, 0, 'normalized')` but that function is _very_ slow in MATLAB and unusable in Octave. So a custom function was written that is probably closer to an approximation of the normalized correlation, but seems to work pretty well.
> 
> [1] https://github.com/proto17/dji_droneid/blob/main/matlab/updated_scripts/find_zc_indices_by_file.m#L118

I tried to use the latest main branch, and used dji air2 droen. Finally, I got a good demodulation effect.
[image]
After,I get the demodulated frame sequence，which consists of 352 characters.  It looks different from the sequence composition in function **create_frame_bytes**. 
It would be great if you could give me some help with frame seq decode.

proto17, 2022-08-19:
Congrats on the demod!  The bytes you are seeing are *all* of the bytes that were demodulated.  The frame length depends on the frame type that was sent by the drone.  To avoid potentially clipping bits that are important, I provide the user with all of the bytes that were demodulated.  If you just want the bytes that matter, change the loop at [1] to use `payload_length` instead of `decoded_bytes.size()`

jianjianliu0918, 2022-08-22:
> Congrats on the demod! The bytes you are seeing are _all_ of the bytes that were demodulated. The frame length depends on the frame type that was sent by the drone. To avoid potentially clipping bits that are important, I provide the user with all of the bytes that were demodulated. If you just want the bytes that matter, change the loop at [1] to use `payload_length` instead of `decoded_bytes.size()`
> 
> [1] https://github.com/proto17/dji_droneid/blob/main/cpp/remove_turbo.cc#L168
Valid! The demodulated sequence can correspond to the fram structure! In addition, I also demodulated the DJI mavic3 drone ID signal, the method proposed by this repo is also effective. 
Impressive work, thanks again for your reply!

closed 2022-08-22 by jianjianliu0918