proto17/dji_droneid #21: Python bindings for utils.h are out of sync

alphafox02, 2022-10-30:
Attempted to Compile the rewrite branch on Ubuntu 22.04. The following issue during Cmake is preventing me from moving forward with testing. 

```
-- Using install prefix: /usr/local
-- Building for version: 1.0.0.0 / 1.0.0
-- No C++ unit tests... skipping
-- Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE) 
-- PYTHON and GRC components are enabled
-- Python checking for pygccxml - found
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython3.10.so
-- Performing Test HAS_FLTO
-- Performing Test HAS_FLTO - Success
-- Found pybind11: /usr/include (found version "2.9.1")
CMake Error at /usr/local/lib/cmake/gnuradio/GrPybind.cmake:239 (message):
  Python bindings for utils.h are out of sync                                                                                                                  
Call Stack (most recent call first):                                                                                                                           
  python/dji_droneid/bindings/CMakeLists.txt:36 (GR_PYBIND_MAKE_OOT) 
```

bkerler, 2022-10-30:
looks like you need to run "gr_modtool bind utils" first

proto17, 2022-10-31:
The rewrite branch is still a work in progress and cannot decode DroneID frames at this time.  As @bkerler pointed out, you need to run `gr_modtool bind utils` in the `gnuradio/gr-droneid` directory.  This was a mistake on the last commit.  Normally you would not have to run the `bind` command.  This has to do with how GNU Radio uses PyBind11.

alphafox02, 2022-10-31:
I see now, I was doing it in the wrong directory. 

Side note, the update branch for 3.10 pumps out the bursts to temp which now I see when I cat them together get good results with Octave. This is with IQ files I previously tried that with, but back then I got no decoded frames. I didn’t check to see if Gnuradio itself was pumping out the same info on the fly.

closed 2022-10-31 by alphafox02