proto17/dji_droneid #26: Can not capture the DroneID signal.

kyz20, 2022-11-10:
I have tried all frequencies you given on Mavic2, but I still can not find the DroneID signal. Can you give me some example picture?
You said the DroneID signal is about 10M bandwidth, and time of duration is 600ms, but I did not see such signal.
Should I turn on the controller or link the phone App to capture the DroneID signal?

kyz20, 2022-11-10:
> I have tried all frequencies you given on Mavic2, but I still can not find the DroneID signal. Can you give me some example picture? You said the DroneID signal is about 10M bandwidth, and time of [...]

Another question, how often does the DroneID signal appear?

lobart, 2022-11-10:
@kyz20 Hi, see my closed issue.
DroneID signal looks like this https://github.com/proto17/dji_droneid/issues/15#issuecomment-1262392201
Signal is produced once per 550-650ms 10-20 times on one frequency carrier and duration is near 600microsecond.

closed 2022-11-29 by kyz20