proto17/dji_droneid #30: AttributeError: module 'gnuradio.fosphor' has no attribute 'glfw_sink_c'

StanleyLex, 2023-01-21:
Hello, there was a problem running correlation_test.grc. My system is ArchLinux. GnuRadio 3.10 from the repository. Gr-fosphor (latest version) is taken from github. Glfw was installed according to the instructions from https://www.osmocom.org/projects/sdr/wiki/Fosphor

I ask for help in solving the problem. Thank you.

`[INFO] [B200] Actually got clock rate 30.000000 MHz.`
`[INFO] [MULTI_USRP]     1) catch time transition at pps edge`
`[INFO] [MULTI_USRP]     2) set times next pps (synchronously)`
`Traceback (most recent call last):`
` File "/home/awerto/correlator_test.py", line 357, in <module>`
`    main()`
`  File "/home/awerto/correlator_test.py", line 335, in main`
`    tb = top_block_cls()`
`  File "/home/awerto/correlator_test.py", line 187, in __init__`
`    self.fosphor_glfw_sink_c_0 = fosphor.glfw_sink_c()`
`AttributeError: module 'gnuradio.fosphor' has no attribute "glfw_sink_c"`

`Done (return code 1)`

proto17, 2023-01-21:
You can disable that block if you'd like.  It's not required for the graph to operate.  It's just there to help you tell that you *should* be getting successful demods.

If you want to see the fosphor display, first make sure that gr-osmosdr and gr-fosphor are installed.  Check to see that your `PYTHONPATH` env variable has the directory that contains gr-fosphor (usually `/usr/local/lib/python3.XX/dist-packages/gnuradio/fosphor` where `XX` is your Python version number.  Could also be in `/usr/lib/python3.XX/dist-packages/gnuradio/fosphor`.  If it's not in either of those locations, then you will have to figure out where the `fosphor` directory is with something like `find /usr -name foshor -type d`.  Once you have the directory nailed down, try running GNU Radio from the terminal with something like:

```PYTHONPATH=$PYTHONPATH:/usr/local/lib/python3.10/dist-packages gnuradio-companion```

Replacing the extra path with whatever the path is that your gr-fosphor modules is in (just up to the `dist-packages` dir, not the hole thing!)

See if that helps.  The issue you are having is related to GNU Radio and gr-fosphor.  Nothing here should be related to the flow graph or gr-droneid.

For fun, you can also go into the glfw source dir, then drill down to `lib/fosphor`, run `make`, then `./main some_random_file` where `some_random_file` is the path to any file (doesn't have to be floating complex).  If a window pops up without errors in the terminal then all is good between fosphor and glfw.  Otherwise you have an error between those two libraries.

Good luck!

StanleyLex, 2023-01-21:
Thanks a lot. It works!