proto17/dji_droneid #32: I don't understand what is mean this results

ShevchenkoVadim, 2023-02-07:
Hi! 
I need your help. 
I received this output. What's mean this results? How I could get drone_id data from this results?

Found burst @ 587225372 / 587225371
7c3ff9feefc6eeebfffc60f5f57db7c1e38febffafbd5417dfffde7fbcSending message with af27952 elements (7e27952) starting at 587225371
f4Found burst @ 587253384 / 587253383
7b34b9bfcfdaec37bfffb9669deedffde8aef6ffbe65b5056dbf7bf76f17beb5fc1d8a8cffef7bafc6f6f7ff6f17fff7fb3033d97fdf4bddc95fffe6acfc1ffe77e5ed273fbfffc52891fbfd979af5efed5f9c95e5bffbfee376737b743779bfffd9fd6d415fdfecd93bc6e7efff297773f7dfa33ebbfddfb665aeed7fde4ee62f7befeeff5a3b74b77958396dfffd99
CRC Check Failed!
Sending message with 27952 elements (27952) starting at 587253383
-----
ZC start index: 9761, est start index: 3025
Sending message with 27952 elements (27952) starting at 591369583
Found burst @ 591397563 / 591397562
Sending message with 27952 elements (27952) starting at 591397562
Found burst @ 591427851 / 591427850
Sending message with 27952 elements (27952) starting at 591427850
Found burst @ 591490373 / 591490372
f68ffffdefff57fc877ff46ac7fffbefaff72eff75b5ddfb7febfefbffd9ff7fbfed6f7cdb7bfd703becdf3b2fbffdb7eb7ee87ff7c5f9fdb9ff5fe7ed0e7ffc9fdffacb8ffffbb6ddf7bfSending message with 27952 elements (27952) starting at 591490372
fddd4fcc7dffff56b6efddf7fbfdfd3fefbbe99fe7bfbde1e9fe77cfb57e6dfffe6ffad3bbfff7febdacaeff7fefe663efbade97dacdfffbf57eb5f8f7fd59c3fffff777767ecfb5ffeb3bb77f9f57f55ddaf5dfbc63ff7a7f3e13bdaf7bcfdbe777aff47e
CRC Check Failed!
Found burst @ 591547527 / 591547526
Sending message with 27952 elements (27952) starting at 591547526
-----
Found burst @ 1190562407 / 1190562406
Sending message with 27952 elements (27952) starting at 1190562406
35d77a616024d9daf4c571e6b6afb784068787d71b363667777a0c1a50f73efe0a7f78fb6095903f7c7161f9ea3b2be847d36be5a7d105c8accf684557a1d457e113969fFound burst @ f2b41ac8471f6c4d6714119062905751 / eae9d2a5541190629056b17ff4
32394d6b6f9c4325b97d8dc9a1758f6de6646fbfeb975493dab0beb416a42c4e771c476be35a2a50d05ece30898deb47892165a6d9f7f48033afbccfa6e917afd7df8889d07f9e37043cdb3f1a526a8bdcfb5a907fef3c09ad
CRC Check Failed!
Sending message with 27952 elements (27952) starting at 1190629056

alphafox02, 2023-02-07:
If you Google gr-droneid YouTube you should find three videos that’ll walk through taking the capture from the air into decoded format. Hope that helps.

ShevchenkoVadim, 2023-02-07:
@alphafox02 Unfortunately I have seen this  video but it didn't help. In this video version of dii_droneid library is so old. Output is different. I'm not sure but as I can see in latest version library gr-droneid could decode frame in realtime.  
Please, tell me how it have to work?

alphafox02, 2023-02-07:
The same steps can apply. I tested it. Still in the latest repo if gr-drone for gr10 there’s still bursts saved to /tmp along with what you’re seeing in the bottom left corner of gnuradio. I take those bursts and combine them into one and run through octave and get the frames. I then run through python script for results. 

I would have to check the real time output vs the info I get from the bursts in temp, but I can say the basic process shown in the few videos works in latest DragonOS (has the newer dji droneid and gr-droneid). Check and see if you’re getting bursts in /tmp and then use those with octave to verify against the live output from gnuradio.

alphafox02, 2023-02-07:
Not dumb at all. Instead of running each burst one by way through the octave script which I have to edit manually, I instead took the burst_* files and then (I’m having a memory lapse) I think just like this is fine..

cat file1 file2 > file3

but do all the burst files. Someone correct me if I’m wrong, then I configure the script file per the video pointing it at the one large file I just made. At the end, if all goes well, it should say found FRAME xxx zzz and have lots info. I then copy and paste the info into a text file per number 3 of 3 videos on YouTube and use the python script inside DragonOS to get the decoded results.

ShevchenkoVadim, 2023-02-08:
@alphafox02 Thank's for your answer it helped me. I have just understood what's my problem. I have Mavic Air 2S and I'm using it for my tests. But I'm not sure that it has Ocusync 2 protocol. What can you tell about this? 
And one more question. What about Mavic 3? I can't get drone_id using this tools?