proto17/dji_droneid #34: Zc sequence identify

jhon-paul-v, 2023-03-15:
Hi, I m interesting about your brute testing to get the root for drone ID zc. Have you tried that on the DJI downlink signal?
My test for downlink zc got a wired result, which has only one root answer for 1022 (I supp￼ose there are two type of zc in one frame). There must be some errors I missed.

proto17, 2023-03-15:
The brute forcing code that currently exists is unlikely to work for the downlink since there are almost certainly a different number of occupied carriers.  My guess is that you'll need to do the following:

- Find the number of occupied carriers in the downlink
- Change the ZC sequence generator to use that number of carriers
- Re-run the tests

I'm not certain this will work, but it might.  It's likely that the downlink also follows the LTE spec in its number of occupied carriers.  If true then you're likely to see one of: 600, 900, 1200, 1800, or 2400 [1] occupied carriers.  I have no idea if they use all of the occupied carriers for the ZC sequence as I haven't had a look.  You can probably look at the spectrogram (waterfall) view in a tool like Baudline to see if they are all being used.

Good luck!

[1] https://faculty.coe.drexel.edu/jwalsh/Gwanmo-Nov11-2.pdf  <--  See slide 4.  Use the bandwidth of the signal to tell how many occupied carriers it should have

jhon-paul-v, 2023-03-16:
Appreciate for your reply,which gives me an new inspiration.
Meanwhile,I still got a problem about the occupied subcarriers.I noticed in function get_data_carrier_indices,a commit says for 2048 fft size,the occupied subcarriers will symmetrical distribute around 1025.Is that common for other number of subcarriers?
For example,I have 2048 fft size with 1000 subcarriers, will 1000subcarriers symmetrical distribute around 1025 fft bin?