proto17/dji_droneid #37: Final step error:  NO DRONE ID

pqoe, 2023-04-20:
Hi, in MATLAB i try a captured file but after a while i get this result
In process_file (line 330) 
Warning: Failed to run the final processing step

proto17, 2023-04-20:
See https://github.com/proto17/dji_droneid/blob/main/matlab/updated_scripts/process_file.m#L325-L329.  It's likely because something went wrong trying to run the turbo decoder.  There should be another error.  Did you provide all of the output?  Looking for something that starts with "[ERROR]".  If you don't have anymore output to show then there might be something wrong with the remove_turbo binary.  Try running `./remove_turbo` with no arguments and see what happens.  If you get a usage statement then all is well, otherwise there's something wrong with the binary.

pqoe, 2023-04-22:
Hi, thanks for replying. I recorded a signal from M600 PRO, around 2414.5 MHz, with a modified Pluto SDR (txco change)
.I'm using matlab on windows. I modified the script process file to add the turbo decoder in same folder of the script such this : 
`turbo_decoder_path = fullfile(this_script_path, filesep, 'remove_turbo.cc');`
In the output i get the demodulated ofdm symbols images but also in matlab output this FRAME : "my current path" is not a recognizable matlab command''
[image]

proto17, 2023-04-23:
I don't think that the M600 sends out DroneID frames on its own [1].  The
signal plots also hint that what you have might not be DroneID.

I don't know if the C++ apps will work in Windows.  You would have needed
to compile the decoder app by hand using something like Cygwin.  But, the
signal you are looking at right now wouldn't decoder anyway :(

[1] https://dronedj.com/2023/04/18/dji-drone-remote-id-module/

pqoe, 2023-04-23:
aaahhh i will find another signal that contains the remote ID, i got access to many drones at work :). I might sound a total noob but i totally do not understand this "You would have needed to compile the decoder app by hand using something like Cygwin." i did not compile anything i just put the cc file in same folder as matlab script, i did install pothos sdr for working with gnu radio but not cygwin. Meanwhile your code looks initimidating but i really appreciate your work it got me hooked

proto17, 2023-04-23:
The .cc files need to be compiled with some form of C++ compiler.  In Windows you could use Cygwin [1], but I don't know for sure if that will work.  All of my development and testing was done in Linux :frowning_face: Using Cygwin is a little bit of a pain if it's the first time you used it.  I don't have a good feel for how easy it will be to build the decoder/encoder apps.  Just placing the .cc files in the MATLAB path will not do anything unfortunately.  Looks like it might be possible to do the compilation from MATLAB [2] but I have no experience with this method

If there are parts of the code that don't make sense just ask!  Thanks for trying it out :smile: 

[1] https://www.cygwin.com/
[2] https://www.mathworks.com/matlabcentral/fileexchange/52848-matlab-support-for-mingw-w64-c-c-compiler

pqoe, 2023-04-24:
@t0rm3nt0r M600 PRO is ocusync i think, whats DL?
@proto17  can i reach you by email? is there anyway to make this work on matlab only, without using gnu radio or g++

proto17, 2023-04-24:
@t0rm3nt0r appreciate the kind words and the knowledge that the code does work in real life :smiley: 

@pqoe i don't normally give out my e-mail.  right now the only options for demod are using the MATLAB with the turbo decoder (includes rate matching) binary.  i recommend using a virtual machine with Ubuntu 22.04.  i'm pretty sure the home version of MATLAB can be installed on 2 machines so you could have the VM be your second.  just keep in mind that you cannot re-home a personal license :(  another alternative is to put a breakpoint at https://github.com/proto17/dji_droneid/blob/main/matlab/updated_scripts/process_file.m#L326, copy the `/tmp/bits` file to a VM, Docker container, or WSL container where you can run the turbo decoder, then copy the output file back to Windows and continue the MATLAB script.  if you think that's a possible solution then i can write up a short guide on how it could be done.  i don't use Windows, so it'll be a bit of a guess.

t0rm3nt0r, 2023-04-25:
@pqoe 
https://www.dji.com/matrice600-pro
"The airframe is equipped with the latest DJI technologies, including the A3 Pro flight controller, Lightbridge 2 HD transmission system, Intelligent Batteries and Battery Management system."

pqoe, 2023-04-25:
@proto17  ill try that, i'm very excited to make it work, if you could write that guide ill be more than thankful !

proto17, 2023-05-13:
@pqoe Give branch 37 a try

pqoe, 2023-05-14:
thank you for your valuable help, i will try this next week. really appreciate your assistance, i'm very gratful 
have nice week :)

pqoe, 2023-05-16:
Hi, im trying the method you posted earlier, but hit a wall during compilation 
[image]
as you can see the turbofec folder is not found, i think it has to do with the g++ compilation  instruction, maybe one of the arguments i dont know

carlosleitao17, 2023-05-16:
Proto17 will correct me if I;m wrong, but I think you need to install libtool.

komalkumar6655, 2023-05-17:
[image]

komalkumar6655, 2023-05-17:
Hi  @proto17 
I have followed as per your suggestion but i am getting these errors
Please can you hep me.
Thank you

proto17, 2023-05-17:
@komalkumar6655 It looks like you ran `wget https://github.com/d-bahr/CRCpp/blob/master/inc/CRC.h` instead of `https://raw.githubusercontent.com/d-bahr/CRCpp/master/inc/CRC.h`.

komalkumar6655, 2023-05-17:
[image]

komalkumar6655, 2023-05-17:
I have run the command you have given but i am getting the above mentioned error like connection timed out.
So ihave run the  wget https://github.com/d-bahr/CRCpp/blob/master/inc/CRC.h

proto17, 2023-05-17:
I think you have a firewall problem as I have no issue with that from home, work, or cell.  The second `wget` command you're running is pulling back an HTML page and not the header file itself.  You can try copy-pasting the contents of the header file into the `cpp` dir by hand.  Just make sure the file is named `CRC.h`

komalkumar6655, 2023-05-17:
Thanks alot for the information in got the application add_turbo .exe and the remove_turbo.exe.

komalkumar6655, 2023-05-17:
Does all DJI drones have the Drone IDs or only specified DJI drones have these Drone IDs.

proto17, 2023-05-17:
Glad it worked!  Only certain DJI drones have DroneID.  I don't have a good list of the ones that do vs don't

komalkumar6655, 2023-05-17:
Thank You 
i have taken the samples file from the github repo: https://github.com/RUB-SysSec/DroneSecurity/tree/public_squash/samples
But still getting the same error as failed to run final processing step

komalkumar6655, 2023-05-17:
FRAME: 'D:\DJI' is not recognized as an internal or external command,
operable program or batch file.

THis error is occuring frequently.

pqoe, 2023-05-18:
I think it worked finally ! i got the 2 files as output, ill recapture an actual RF iq from my pluto sdr then work the matlab code on it, ill let you know later, thanks again !