proto17/dji_droneid #38: Detect DroneID burst without "gr-fosphor"

jaehoon1001, 2023-05-11:
Hi, I have a few questions about this.  
  
I want to capture DroneID burst via HackRF (not decoding, just burst detection). You used GNU Radio's `gr-fosphor`, but I can't use this module due to environmental problems (I have some error about gr-fosphor 😥). 
  
So I want to capture it using a different visual module. Can I detect DroneID burst using "QT GUI Sink" block or other blocks? (or using a radio receiver such as gqrx)

proto17, 2023-05-12:
gr-foshor is used because it can show you things that the normal QT Frequency Sink cannot.  Specifically bursting signals.  The update rate for the FFT module might be fast enough at 15.36 MSPS but your monitor is not up to the task of displaying those FFT results back.  Assuming at 1024 FFT, that would be 15,000 frames per second that you would have to display in order to never miss a burst.

Short answer: It's unlikely that the QT GUI Frequency Sink will be of much help.  Most modern computers can make use of gr-fosphor, you just might have to use the software implementation.

You can still use the GNU Radio flow graph, you just won't know if the frequency you are currently on is active or not.  There's another ticket here that aims to help with that problem by frequency hopping your SDR along to all of the channels.

alphafox02, 2023-05-13:
You can use gr-fosphor pretty well via cpu. An example when using DragonOS,
https://sourceforge.net/p/dragonos-focal/discussion/general/thread/fe60025587/
I’ve used it surprisingly well on both Intel CPU and the steam decks amd cpu.

jaehoon1001, 2023-05-13:
@proto17 Thank you! Your answer was very helpful. I was able to capture a signal that was supposed to be DroneID. I used the method of hopping all the frequencies you suggested :) This is my capture signal.
[image]

jaehoon1001, 2023-05-13:
@alphafox02 I only used Ubuntu, but I will also use DragonOS that you told me about. Thank you for letting me know more options

closed 2023-05-13 by jaehoon1001