proto17/dji_droneid #41: This DJI Drone ID Not Working for Phantom 4 Pro v2.0 Data

KomalKumar96, 2023-06-13:
Hello @proto17 
We have recorded the DJI Phantom 4 Pro v2.0 Data with our IQ Recording system. 
We have reached till the CPP calling function in the matlab code.
But the CRC error is coming for all the frames that are decoded.

Is there any possible solution for this. Or is there any packet structure for the DJI Phantom 4 pro v2.0

proto17, 2023-06-19:
Apologies for the delayed response.  I don't know if that drone supports DroneID.  The message you sent me hints that the correlation scores are very low which likely means that either what you have recorded is not DroneID, or something in the protocol changed for your drone.  I don't have any of those drones on hand to test with, so I can't personally verify.  If you post a screenshot of the waterfall plot of the signal I can probably tell you if what you have definitely isn't what I know DroneID to be.

waseem-arshad94, 2023-12-01:
@proto17 Dear Sir;
I am getting this error when I started with DJI Phantom 4 Pro V.2 CRC Check failed. Following are my outputs.
[image]
[image]
[image]

Thanks