proto17/dji_droneid #43: Mavic 3 DroneID descrambling

arglas864, 2023-07-25:
Hello everyone,
I'm trying to decode the Mavic 3 DroneID, but there's something I don't understand.
In your guide @proto17 you say that each burst contains 4 symbols with ZC sequences, and the remaining 6 symbols are data symbols, the same number as in the DJI Mini 2. However, for the Mini 2 the first data symbol is discarded before the descrambling, so in total there are 7 data symbols and 2 ZC symbols. 
If I remove the first data symbol in the Mavic 3 droneID burst I end up with 5 data symbols, and the descrambling step fails because it needs 6 symbols. 
So, my doubt is: for the Mavic 3, do I have to add the first data symbol for the descrambling, or the descrambling process isn't the same as the Mini 2?
Thank you

proto17, 2023-09-03:
This repo assumes that the first OFDM symbol (first symbol in the 9-symbol case) is **not** present as it's not important.  So you shouldn't have to remove any OFDM symbols.  Each burst only contains 2 OFDM symbols with ZC sequences.  Could you show me where I mistakenly wrote 4?  I'd like to fix that.

In the 9 OFDM symbol case there are really just 6 data symbols as the first "data" symbol is a duplicate of the first.  That's why it's safe to ignore the first symbol.  I'm not sure that the Mavic 3 uses the same signal structure, so I can't really tell you what should or should not change =\

arglas864, 2023-09-05:
Thanks for your answer. I was referring to this guide: https://github.com/proto17/dji_droneid/wiki/DJI-Mavic-3-DroneID-Analysis .
Anyway, I was able to solve the problem: I was simply looking at the wrong bursts. It seems that the Mavic 3 trasmits two types of message in addition to the video downlink. One is the standard Drone ID which is similar to Mini 2's, the other one is this strange message using 4 ZC sequences. 
With the correct bursts the decoding process went ok.

closed 2023-09-05 by arglas864