proto17/dji_droneid #44: Low Correlation Scores

jianzhitaiyouqule, 2023-09-03:
I used a sample file from [https://github.com/RUB-SysSec/DroneSecurity](url),and then i changed the paramters of process_file.m like this:
[image]
i got low Correlation Scores.

Could someone help me?

jianzhitaiyouqule, 2023-09-03:
Low Correlation Scores
[image]

jianzhitaiyouqule, 2023-09-03:
[image]
Is this DroneId?

proto17, 2023-09-03:
The `mavic_air_2` file appears to be sampled at 50 MSPS which is not a valid rate for `process_file.m` (must be a multiple of 15.36 MSPS).  Additionally, just looking at the signal in Baudline, it appears that there is a 2 MHz offset assuming a 50 MSPS sample rate.  So, you'll need to resample the file from 50 MSPS to 15.36 MSPS or 30.72 MSPS, then change the `file_freq_offset` parameter to `2e6` and set `file_sample_rate` to the resampled rate.  I resampled the file with the script below, and then fed that resampled file into `process_file.m` using a sample rate of `30.72e6` and frequency offset of `-2e6` (to undo the 2 MHz offset in the file).  From there I saw good constellation points, but a failed CRC.  The demodulated bytes look good, but the CRC does not pass.  I haven't put it through any kind of parser to see if the data is actually valid.

There are multiple bursts present, but you would need to resample to 61.44 MSPS before feeding into `process_file.m` to be able to demodulate those using the `file_freq_offset` parameter.

```
file = '/tmp/mavic_air_2';
samples = read_complex(file, 0, Inf, 'single');

original_sample_rate = 50e6;
target_sample_rate = 30.72e6;

resampled = resample(samples, target_sample_rate, original_sample_rate);

figure(1);
subplot(3, 1, 1); plot(10 * log10(abs(fftshift(fft(samples)).^2))); title('Original File');
subplot(3, 1, 2); plot(10 * log10(abs(fftshift(fft(resampled)).^2))); title('Resampled File');

% Just appending `.resampled` to the file name
output_name = [file, '.resampled'];
handle = fopen(output_name, "w");
fwrite(handle, reshape([real(resampled), imag(resampled)].', [], 1), 'single');
fclose(handle);

subplot(3, 1, 3); plot(10 * log10(abs(fftshift(fft(read_complex(output_name, 0, Inf, 'single'))).^2)));
title('Read Back Resampled File');
```

jianzhitaiyouqule, 2023-09-06:
I resampled the file from 50 MSPS to 30.72 MSPS, just as you said，there is a CRC error.
[image]
Then i resampled the file to 61.44 MSPS, I got two different frames.
[image]

I'm trying to decode it to get the information by using dji_decode.py from dragon OS. I found it from the comments.

Thank you very much for your reply, you made me realize that I overlooked a lot of details,really appreciate your assistance.😀

closed 2023-10-07 by jianzhitaiyouqule