proto17/dji_droneid #46: Signal Detection

SpongeBobaa, 2024-01-15:
The following is the signal I collected using USRP. May I ask if this is a DroneID burst signal?

[image]

wanhongwu1998, 2024-05-27:
yes! it is

yous9990, 2024-05-27:
did the decoding changed?

proto17, 2024-05-27:
It does look correct in terms of number of OFDM symbols and location/structure of the ZC sequences. I can't tell if the sample rate is correct from the image though, so make sure that you're capturing at 15.36 MSPS or properly resampling in the `process_file.m` script

2hang0, 2025-03-02:
> 以下是我使用 USRP 收集的信号。请问这是不是 DroneID 突发信号？
> 
> [image]

Hello, could you send me a copy of the signal file you collected during your testing (IQ file 、fc32file or raw file)? I would like to use Python code to test whether I can decode the signal, but I am unable to capture any signal.Please.

King-Of-Knights, 2025-08-17:
> What is the result of the successful operation of the project? For example, what information will be printed out of the command line window, if you can, please take a screenshot to show me. I always end up like this. [image]

会给你一串十六进制码，并且没有CRC报错
[image]