proto17/dji_droneid #49: gnuradio import droneid

gettyhub, 2024-05-13:
Had to clone turbofec (https://github.com/ttsou/turbofec) & CRCpp (https://github.com/d-bahr/CRCpp) in order to build on Ubuntu 22.04 with gnuradio 3.10.

However when I go into the droneid_test.grc, it comes up with an error about importing droneid.

proto17, 2024-05-27:
Could you share the exact version of GNU Radio that you're using and whether or not you built it from source? That would help me try to recreate the issue.

Could you also share the error that you're seeing?

gettyhub, 2024-05-27:
> Could you share the exact version of GNU Radio that you're using and whether or not you built it from source? That would help me try to recreate the issue.
> 
> Could you also share the error that you're seeing?

Gnuradio 3.10.1.1-2 and the import error is in gnuradio-companion, in the import block.  Import droneid from gnuradio.

proto17, 2024-05-27:
I think you may have been bitten by Python not looking in `/usr/local/lib/python<VERSION>/dist-packages`. If you run `find /usr/local/lib/python3.* -name droneid -type d` you should get a single directory where the `droneid` package lives. This all assumes that you didn't do any trickery when installing the module.

As an example, if the command above returned `/usr/local/lib/python3.10/dist-packages/gnuradio/droneid` then you could launch GNU Radio Companion from the command line with: `PYTHONPATH=$PYTHONPATH:/usr/local/lib/python3.10/dist-packages gnuradio-companion` and things should be happy. The `$PYTHONPATH` up front is there in case the `PYTHONPATH` environment variable has already had paths added to it.

Glad to hear a reboot seems to have solved the problem!