proto17/dji_droneid #50: Ocusync 4 cannot be resolved

wanhongwu1998, 2024-05-27:
Hello, thank you very much for your sharing. Our research has found that this burst does not exist in LightBrige technology, so it is impossible to parse such a drone. In addition, although the burst can be found in Ocusync 4 technology and the constellation image can be demodulated successfully, the signal of the drone cannot be solved, as it seems to be encrypted. I wonder if you have encountered the same problem in the future.

proto17, 2024-05-27:
I have not done any work with newer drones. If you have a collect that you're willing to share I would be happy to have a look (keeping in mind your physical location could be present in the collect!), but the only drone I have is a Mini 2. If the constellation comes out clean and you get data back then I would suggest looking at that data with a hex editor to see if there are any strings present. IIRC the serial number of the drone shows up in the demodulated output. If you see that serial number in the output then you can be pretty sure that the link isn't encrypted. It could be that the CRC changed, or that the format of the demodulated output changed. But it's definitely still possible that the link has been encrypted. There's been lots of calls for that to happen. 

It could also be that you have a really noisy collect that's failing to properly demodulate. The `process_file.m` script doesn't output the frame hex if the CRC fails. You can change this by commenting out [1]. Doing so will output the hex for all frames even if they are not valid according to the CRC. 

Good luck!

[1] https://github.com/proto17/dji_droneid/blob/main/cpp/remove_turbo.cc#L156

wanhongwu1998, 2024-05-29:
Thank you for your reply. I'm sorry that it is not convenient to export the data I collected to you! To be sure, CRC was successful, so I'm more confident that the format of the demodulated output changed or was encrypted.

jianjianliu0918, 2024-06-04:
> Thank you for your reply. I'm sorry that it is not convenient to export the data I collected to you! To be sure, CRC was successful, so I'm more confident that the format of the demodulated output changed or was encrypted.

DJI officially said it was encrypted, but recently a company claimed it had successfully decrypted it.

wanhongwu1998, 2024-06-14:
> DJI officially said it was encrypted, but recently a company claimed it had successfully decrypted it.

I doubt they can parse encrypted data. If encrypted data can still be parsed, then DJI's engineers are useless, and of course it doesn't rule out that they have superior abilities or have some special deal with DJI.

ozymandiasmek, 2024-11-19:
> Hello, thank you very much for your sharing. Our research has found that this burst does not exist in LightBrige technology, so it is impossible to parse such a drone. In addition, although the [...]

Did you revolve ocusync 3.0 ?

alphafox02, 2024-11-19:
Even if the signal can’t be decrypted, there’s still value in knowing it exists. If a drone signal is within range of your receiver, and you deploy additional receivers positioned at strategic distances, maybe you could triangulate the signal’s location based on the strength detected by each sensor.

This assumes the signal can still be identified, even if it’s encrypted. I believe the AntSDR e200 firmware works similarly—it detects encrypted signals and flags them as such, without decoding the contents.

1264763882, 2025-01-07:
您好 我想问一下ocusync4是通过时间加密的吗 我的qpsk簇已经解析的比较好了 但解码部分误码严重

jianjianliu0918, 2025-01-07:
> 您好 我想问一下ocusync4是通过时间加密的吗 我的qpsk簇已经解析的比较好了 但解码部分误码严重

国标SM2加密的，不用想着破解了

1264763882, 2025-01-07:
我观察dji协议无人机数据解扰送进去的数据去除zc序列后固定从4148位置送入1412个bit码元 解码位置是否有变化

RenDJunyu, 2025-03-24:
> 您好 我想问一下ocusync4是通过时间加密的吗 我的qpsk簇已经解析的比较好了 但解码部分误码严重

DJI的droneid应该是可以解调的，但是图传和跳频应该是加密的

Umairnxr, 2025-09-09:
Hello, I am working on the Ocu4 Drone ID decoding. is there anyone who can help me out. I have a mini 4 pro.

Sanafan, 2026-03-20:
anyone able to sell decryption key or server?

luyii-code-1, 2026-09-11:
A small update on O4: I have been working on a reproducible O2/O4 DroneID receiver and packet-analysis project, and the encryption structure is now considerably clearer than when this issue was originally opened.

Repository:
https://github.com/luyii-code-1/dji-ocusync-droneid-research

Cross-validation thread:
https://github.com/luyii-code-1/dji-ocusync-droneid-research/issues/1

For DJI Mini 5 Pro captures I can recover O4 AA/87 logical packets with both CRC24A and DJI CRC16 passing.

The currently verified chain is:

```text
AA
  -> SM2-compatible C1 || C3 || C2
  -> 16-byte note

87
  -> AES-128-CTR
  -> SN / UUID / drone / pilot / home telemetry
```

The AA key-material region has a 64 + 32 + 16 byte layout matching raw-point SM2 C1||C3||C2, and C1 validates on the SM2 curve across multiple sessions.

The 16-byte `note` from the paired AA is the AES-128 key for 87. With the correct note, sequential 87 packets decrypt into coherent telemetry using:

`IV = nonce8 || 0x00*8`

AA and same-session 87 packets also share the same 4-byte hashcode.

So the remaining hard part is now specifically `AA -> note`, likely requiring the corresponding SM2 private key or an equivalent decryptor, rather than understanding the 87 ciphertext format itself.

I would be very interested in independent O4 captures from other models/firmware for cross-validation.

Hellikandra, 2026-09-24:
> A small update on O4: I have been working on a reproducible O2/O4 DroneID receiver and packet-analysis project, and the encryption structure is now considerably clearer than when this issue was originally opened.
> 
> Repository: https://github.com/luyii-code-1/dji-ocusync-droneid-research
> 
> Cross-validation thread: [luyii-code-1/dji-ocusync-droneid-research#1](https://github.com/luyii-code-1/dji-ocusync-droneid-research/issues/1)
> 
> For DJI Mini 5 Pro captures I can recover O4 AA/87 logical packets with both CRC24A and DJI CRC16 passing.
> 
> The currently verified chain is:
> 
> ```
> AA
>   -> SM2-compatible C1 || C3 || C2
>   -> 16-byte note
> 
> 87
>   -> AES-128-CTR
>   -> SN / UUID / drone / pilot / home telemetry
> ```
> 
> The AA key-material region has a 64 + 32 + 16 byte layout matching raw-point SM2 C1||C3||C2, and C1 validates on the SM2 curve across multiple sessions.
> 
> The 16-byte `note` from the paired AA is the AES-128 key for 87. With the correct note, sequential 87 packets decrypt into coherent telemetry using:
> 
> `IV = nonce8 || 0x00*8`
> 
> AA and same-session 87 packets also share the same 4-byte hashcode.
> 
> So the remaining hard part is now specifically `AA -> note`, likely requiring the corresponding SM2 private key or an equivalent decryptor, rather than understanding the 87 ciphertext format itself.
> 
> I would be very interested in independent O4 captures from other models/firmware for cross-validation.

Independent cross‑validation from a separate O3/O4 analysis project — I can confirm your AA/87 model byte‑for‑byte from a different codebase:

CRYP 0xAA = SM2 point C1‖C3‖C2 = 64+32+16 = 112 bytes, 4‑byte hashcode
INFP 0x87 = AES‑128‑CTR, IV = nonce8 ‖ 0x00×8, same 4‑byte hashcode as its CRYP
CRYP↔INFP linked by that hashcode; keyed‑INFP decrypt reproduces coherent SN/lat/lon/alt/pilot/home

Our detector/parser uses the identical constants and passes both CRC‑24A and the DJI CRC‑16, so this is now reproduced by two independent implementations. Happy to send independent O4 captures from other models (Avata 2 / Goggles 3) for cross‑firmware validation — I can export a JSON packet dump (CRYP split into C1/C3/C2, INFP nonce+ciphertext, hashcode, CRC status) that should drop straight into your pipeline.

One correction that I think saves effort, re: "the material to decrypt the session key is embedded in both the drone and the RC" — for the SM2 layer that isn't the case, and it's worth being precise because it changes where to look:

CRYP is the session key encrypted with DJI's SM2 public key. Unwrapping it needs the SM2 private key, which by design is not on the drone or the RC — the drone only holds the public key (to encrypt), and DroneID is a broadcast for remote‑ID, so the RC never decrypts CRYP at all (it already has telemetry over the control link). So firmware strings/RE will surface DJI's public key and the AES‑CTR/hashcode code, but not a CRYP‑decrypting private key — it isn't there to find.
The only on‑device route is the other end: the aircraft generated the 16‑byte session key itself, so it holds the plaintext key transiently in RAM before wrapping it. Recovering a session's telemetry therefore means lifting that 16‑byte key from the aircraft's own memory (a root shell / hardware access on the drone SoC), not decrypting CRYP. That's the realistic (if heavy) path; the AA → note step stays walled by SM2 (ECDLP, ~2¹²⁸) without DJI's private key.
Also worth separating: the DroneID broadcast session key (SM2‑wrapped, per‑flight) is distinct from the control‑link binding session key (the OcuSync data channel). Conflating them sends you looking for the wrong key in the wrong place.

Net: the 87 format is fully understood (agreed), and AA → note is a hard asymmetric wall — reachable only via DJI's private key or the aircraft's RAM, not via RC/firmware key material. Glad to contribute the cross‑model captures whenever useful.

Sanafan, 2026-09-24:
I have many drones... 

M400
M30T
4x M4TD
M4E
Mavic 4 Pro

How to contribute?

luyii-code-1, 2026-09-24:
The plane isn't necessary, mainly the SDR equipment

Sanafan, 2026-09-24:
> The plane isn't necessary, mainly the SDR equipment
> 
> 
> 
> 
> 
> 
> 
> ---Original---
> 
> From: ***@***.***&gt;
> 
> Date: Fri, Sep 25, 2026 01:09 AM
> 
> To: ***@***.***&gt;;
> 
> Cc: ***@***.******@***.***&gt;;
> 
> 
> 
> 
> 
> 
> 
>  
> 
> I have many drones...
> 
>  
> 
> M400
> 
>  M30T
> 
>  4x M4TD
> 
>  M4E
> 
>  Mavic 4 Pro
> 
>  
> 
> How to contribute?

I also have an ant sdr e200

luyii-code-1, 2026-09-24:
Maybe we could try to get some "AA/80" packets. Even though we've pretty much confirmed the encryption method now, cross-validation can never hurt

  

 
 
 
 
 
 
 
Sanafan left a comment (#50)

Hellikandra, 2026-09-24:
Here is maybe an idea : 

**Goal:** build a multi‑model / multi‑firmware dataset of DJI **OcuSync 4** DroneID
packets so the **AA/87 structure** can be independently confirmed across the whole O4
line (Mini 4, Air 3/3S, Mavic 4, Avata 2, Matrice 4/30/300‑class, …).

This is **receive‑only, no flight required, no keys required.** You are only recording
and parsing the *structure* of the public remote‑ID broadcast — the telemetry payload
stays encrypted (that's expected; see "What this does / doesn't do").

Model confirmed so far (Mini 5 Pro, independent decoders agree):

```
AA (CRYP) -> SM2 point C1||C3||C2 = 64+32+16 = 112 bytes, + 4‑byte hashcode
87 (INFP) -> AES‑128‑CTR, IV = nonce8 || 0x00*8, same 4‑byte hashcode as its AA
```

---

## TL;DR

```bash
# EITHER just share your raw capture:
#   grab.s16.iq  +  the metadata block below      (we/others demodulate it)
#
# OR, if you can demodulate to bytes, run the ONE self-contained file (no dependencies):
python3 o4_droneid_parse.py demod.bin --json o4_pkts.json --model "DJI Avata 2" --firmware "V01.00.0100"
# then share o4_pkts.json
```

---

## Requirements

- **An SDR** that reaches 2.4 and 5.8 GHz: ADALM‑Pluto, **AntSDR E200**, USRP B2xx, etc.
- **Python 3** — nothing else for the parser.
- **`o4_droneid_parse.py`** — a **single, self-contained** file (stdlib only, no numpy, no other
  project files). This is the only file you need to produce the standardized dump; it is included
  alongside this README.

> The maintainers use a larger `sdr_droneid_o4.py` (adds an IQ *detector* and a keyed-decrypt test,
> which need `numpy`/`cryptography`). Contributors don't need it — `o4_droneid_parse.py` produces the
> identical JSON from any demodulated byte stream.

---

## Two ways to contribute (pick your level)

### Level 1 — raw IQ grab (easiest; anyone with an SDR)

Just record the drone's DroneID emission and share the file + metadata. We/others demodulate it.

1. Power the drone (a GPS lock helps populate the telemetry, but a bench capture is fine).
2. Record **~5 s** of IQ at **15.36 MS/s** on the DroneID band (2.4 **and** 5.8 GHz if you can):
   - Pluto/AntSDR example rate: `15.36e6`. Centre on the operating channel; DroneID hops, so a
     wide grab (AntSDR's larger bandwidth is ideal) catches more.
   - Save as interleaved **int16 I,Q** (`.s16.iq`) — the format the tools read.
3. Share `grab.s16.iq` (or a trimmed clip) **plus the metadata block below**. That's it —
   no tool needed at your end for Level 1.

### Level 2 — parsed packet dump (if you have a working O4 receiver)

If you already demodulate O4 DroneID to bytes (e.g. with the luyii receiver, or your own),
run the single self‑contained parser to produce the standardized JSON:

```bash
python3 o4_droneid_parse.py demod.bin --json o4_pkts.json --model "..." --firmware "..."
```

`demod.bin` = the demodulated DroneID byte stream (the 0x55‑framed logical packets).
[... 132 lines cut ...]
    ap.add_argument("--firmware", default="", help="firmware version (metadata)")
    a = ap.parse_args()
    pkts = parse_o4_packets(open(a.bin, "rb").read())
    for p in pkts:
        print(p)
    n_cryp = sum(1 for p in pkts if p.get("type") == PKT_CRYP)
    n_infp = sum(1 for p in pkts if p.get("type") == PKT_INFP)
    n_link = sum(1 for p in pkts if p.get("linked_cryp"))
    print(f"\n{len(pkts)} CRC-valid packets: {n_cryp} CRYP, {n_infp} INFP, {n_link} linked")
    if not pkts:
        print("(no CRC-valid O4 packets — check the demod / byte alignment)")
    if a.json:
        meta = dict(source=a.bin, tool="o4_droneid_parse", model=a.model, firmware=a.firmware,
                    constants=dict(CRYP=hex(PKT_CRYP), INFP=hex(PKT_INFP), sm2_point_len=SM2_POINT_LEN,
                                   layout="C1(64)||C3(32)||C2(16)", hashcode_len=HASHCODE_LEN,
                                   infp_iv="nonce8||0x00*8", cipher="AES-128-CTR",
                                   crc8_seed="0x77", crc16_seed="0x3692"),
                    n_packets=len(pkts), n_cryp=n_cryp, n_infp=n_infp, n_linked=n_link, packets=pkts)
        json.dump(meta, open(a.json, "w", encoding="utf-8"), indent=2)
        print(f"[json] wrote {a.json}")
    return 0 if pkts else 1

if __name__ == "__main__":
    sys.exit(main())
```

wanhongwu1998, 2026-09-28:
Hi,

I’m very glad to receive your message, and I’m happy to exchange information with you.

I’ve attached two files containing packets that have already passed the CRC checks. I believe they were captured from a DJI Mavic 4 Pro, although I don’t clearly remember the exact aircraft model and firmware version anymore. I hope these data can still be useful for your analysis.

I also tried to work on local/offline parsing of O4 DroneID before. I did some initial research and experiments, but I had other work to focus on at the time, so I eventually put it aside.

Recently, I heard that some people may already be able to perform partial local parsing of O4 and O4+, although the results still cannot achieve the same level of performance or completeness as the server-side decryption. That makes me very curious.

After I finish my current stage of work, I’m also planning to spend some time looking into local O4 parsing again. I’d be very happy to stay in touch and exchange any findings or useful data.

Best regards,
Wanhong

87 10 49 4E 46 50 A7 69 6A 19 50 75 55 C4 5E 7D A0 90 6D 00 1A A5 5C D9 D7 89 1B EB C7 BB 5B E7 98 36 CA F9 94 1C A9 99 B8 C0 18 38 51 97 C6 B5 48 46 01 50 97 1F D9 A5 4D 71 BB 24 B7 BE 7F FE 61 1E CD 99 27 75 61 92 5E E4 F0 E3 A4 15 48 48 78 BF 6D A8 5A 8D 42 E0 0B 40 56 3C 31 01 51 45 79 BF 42 8B 05 95 E1 C5 AC A2 C1 D7 FF 8A 7A 31 2A CC A0 B2 B0 6C 25 01 9B 4B 49 FD 64 5B B7 61 02 F0 20 00 57 02 3A C5 35 94 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 7A 38 BB D3 0F 76 EC
A7 11 49 4E 46 50 A7 69 6A 19 88 41 A6 02 83 2F 78 E1 8D 00 2A D4 23 C5 AF 82 D9 D1 2B C4 39 88 AF 82 F5 AE B5 0A 5C 45 DC C3 F8 98 5C 4C 0B 66 65 5A F1 90 43 34 68 19 87 FC 48 7F 63 9A 81 C2 36 25 50 63 8A 02 59 AF CF 5E 37 E1 E9 B2 91 75 A3 26 A9 90 17 25 F5 97 C7 B4 DD 42 DE B4 D2 CF 5C 01 34 F2 E1 27 CB 55 D7 DD C7 B0 A7 E5 CE 82 C8 EB 35 17 61 5B 8A 42 0A 10 DD 69 FF 33 53 10 0E F0 89 07 18 1F DB EA 06 48 3B 4A 95 4D 98 73 03 A6 2A CD D7 BA 51 FF 81 74 B6 B8 31 83 06 BC A6 4F 84 3C 00 00 00 00 7B 1A 00 00 00 9B BD 5A
87 10 49 4E 46 50 A7 69 6A 19 68 BD CD 73 69 70 07 1E 6D 00 B8 66 6F 91 5F 92 84 83 DC A1 40 25 4B 30 53 F0 3E A8 F7 8F F7 E2 CE 0E 0A 9C 56 94 96 0A AA CF E9 8C 48 66 03 0C 95 CE 06 F7 9E CB A7 AF 8E 63 72 17 92 FE F5 2B 44 D7 E7 1D AC 69 79 66 BB CA 92 57 A2 D5 63 F0 97 11 1A 88 8C 00 86 BA B7 C5 DF C8 95 92 C7 51 D9 EE C2 CF 19 99 B8 C3 75 A3 24 92 6C BA D1 A0 BD EE B7 86 B2 19 33 7E 20 00 37 EF 83 36 24 57 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 7A 38 BB D3 52 CD 3A
87 10 49 4E 46 50 A7 69 6A 19 94 B2 1B 61 5B 7D FF 89 6D 00 AD 43 0F DC 84 38 DD 8D FB 05 A3 44 7A 94 0B 5B C8 EE 38 B5 FA 7D 98 36 CC AD E5 BA 39 6D AE 53 E5 B7 67 0C A5 51 A0 87 CD B2 82 7A 4C D5 A5 30 04 4F E0 40 62 66 73 3C D4 8E 0D 10 D9 B4 8F D5 07 03 5C C4 7C 3C C4 84 3A 72 3A 71 8D 68 BF B8 77 48 5E 1D CA 6E FA 53 37 5A AE E6 1A 55 07 5A 88 81 80 F3 71 61 56 6F 7E 23 18 26 CA E9 20 00 10 05 36 BA 17 84 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 7A 38 BB D3 4A 4D C1
87 10 49 4E 46 50 A7 69 6A 19 F4 61 89 CB EE E9 BB E5 6D 00 2E 33 65 6C 77 A8 CB 2C EB 07 BE 18 24 4F 63 CB 6F 2A 9D 54 3C AF 6F DE 0E 29 71 A2 04 C4 33 D6 D0 33 7B 7C A6 8E A1 24 A3 B8 77 AC 86 35 D1 C8 C9 54 9D 95 94 3C 98 67 F1 5F 08 5B 2B 27 99 62 83 36 5E 9C 20 CC 6F B2 09 93 1B C0 4D 86 BE 89 43 B2 22 EA C1 E7 0E 52 AB 23 18 15 87 7A 20 82 D5 79 59 A6 29 4F E1 CA B2 1E E9 54 CE 44 20 00 4A 80 67 A9 45 37 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 7A 38 BB D3 89 EC 4B
87 10 49 4E 46 50 A7 69 6A 19 20 56 D7 B9 E1 F7 B3 50 6D 00 37 3A 8C 8E F6 A3 32 8B 60 0E 20 3C 43 94 1E F9 3D 65 4E 50 45 13 77 1C 56 CE 10 8E 0B 1B 76 C5 C7 AC 0D 47 CC D0 21 37 CC FE B4 B2 5F 8F BE B8 D6 46 B4 52 F2 89 47 83 19 6D DB C3 29 98 AE 13 EE 93 9A CA 53 10 7D 75 EE 88 29 B2 06 BC 3F C3 81 36 22 40 58 2A E9 8D 2D 62 71 D9 CE 9B 06 1B C2 DB 40 57 9F 0F D0 33 3C 01 07 37 A6 B0 20 00 06 E1 97 6F 01 57 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 7A 38 BB D3 70 FA D5
87 10 49 4E 46 50 A7 69 6A 19 78 42 73 93 C5 12 A2 27 6D 00 28 CB FF 4D 1D 8D 4B CF EC 15 8D FD 37 48 5F 97 77 B8 DB 34 F5 76 00 FC A8 A9 96 E2 DB 47 34 12 FF EB AA D2 81 D2 5F 0C 40 33 47 51 DF C5 D0 FA E7 2D DD 6A 9E 0F 91 D5 38 91 85 3A 5C E9 DD 75 4A 56 4D 99 F9 B5 4A D7 E7 3E EC 52 C0 E9 5E B0 22 12 E5 A0 F5 BA F7 05 F5 D1 6F 72 0D 8E 7C A0 9B 96 C1 22 C8 8E 3C F1 DD 1C 15 22 49 87 20 00 18 97 C1 0D 4C 49 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 7A 38 BB D3 15 6C 19
87 10 49 4E 46 50 A7 69 6A 19 80 04 45 23 74 63 6F AC 6D 00 94 75 2B BB 18 D1 2F 65 FB 57 91 CB B7 B7 11 EB 99 66 A2 52 9B 9E 87 0E 02 EB 02 32 7A 45 62 F7 AF 2C 00 B5 9E A4 67 CA DC 2B 6B EF FA E6 C3 15 24 EC 2A 9F A8 5B 8B 7D BB 0E B0 AC 9C BD 0F 2F 8F 54 CE 7D 6F F0 42 E4 AF 14 52 A4 11 E6 75 EE 30 0E 74 9A 4A AE 20 EF 40 33 DD 40 FD 49 C5 3D AF 17 B5 E2 C4 8B 0B 66 B7 FE B8 03 D2 0B 20 00 D0 3F 0A A2 96 9C 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 7A 38 BB D3 53 64 79
87 10 49 4E 46 50 A7 69 6A 19 0C A8 02 7B F9 DC 22 73 6D 00 15 F7 5B B5 13 39 24 5A 80 D8 00 FE B7 E9 2A C4 BA 4C A0 0B 26 E3 A2 BE CD 8C AD DA 8F 61 1D 39 B5 77 0B 74 22 39 30 09 87 C8 4D 04 F0 9B CF 2A C7 0C D7 E4 D3 2D C6 8E C8 58 C7 E9 95 [... cut ...] 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 7A 38 BB D3 7A 20 7B
AA 13 43 52 59 50 A7 69 6A 19 57 BE 26 C1 D9 89 5F EF E2 DF 78 41 A5 A2 12 0C 43 A6 F7 D2 41 36 34 6F A2 DC 05 62 A0 2B 12 E7 99 36 A1 42 98 D8 F5 22 C1 05 97 5F A0 18 1E F9 27 C9 53 D5 D7 AF B1 D2 78 6C 82 98 42 2B 77 77 45 38 0C 63 21 F1 5B C7 88 3B 9E 7D F0 C8 2E A3 38 E2 B2 F1 3A EA D9 4A 8B 15 B7 F6 58 61 BD D7 62 BD ED ED D2 D5 B8 E8 E4 59 7B 47 34 E3 5E 53 13 A3 41 DB D4 5E 41 19 20 00 DA 33 40

wanhongwu1998, 2026-09-28:
Hi,

I’m very glad to receive your message, and I’m happy to exchange information with you.

I’ve attached two files containing packets that have already passed the CRC checks. I believe they were captured from a DJI Mavic 4 Pro, although I don’t clearly remember the exact aircraft model and firmware version anymore. I hope these data can still be useful for your analysis.

I also tried to work on local/offline parsing of O4 DroneID before. I did some initial research and experiments, but I had other work to focus on at the time, so I eventually put it aside.

Recently, I heard that some people may already be able to perform partial local parsing of O4 and O4+, although the results still cannot achieve the same level of performance or completeness as the server-side decryption. That makes me very curious.

After I finish my current stage of work, I’m also planning to spend some time looking into local O4 parsing again. I’d be very happy to stay in touch and exchange any findings or useful data.

Best regards,
Wanhong

AA 13 43 52 59 50 A7 69 6A 19 57 BE 26 C1 D9 89 5F EF E2 DF 78 41 A5 A2 12 0C 43 A6 F7 D2 41 36 34 6F A2 DC 05 62 A0 2B 12 E7 99 36 A1 42 98 D8 F5 22 C1 05 97 5F A0 18 1E F9 27 C9 53 D5 D7 AF B1 D2 78 6C 82 98 42 2B 77 77 45 38 0C 63 21 F1 5B C7 88 3B 9E 7D F0 C8 2E A3 38 E2 B2 F1 3A EA D9 4A 8B 15 B7 F6 58 61 BD D7 62 BD ED ED D2 D5 B8 E8 E4 59 7B 47 34 E3 5E 53 13 A3 41 DB D4 5E 41 19 20 00 DA 33 40 2F 0D 08 AF 14 45 2D 94 23 7F 30 01 EF E2 63 E3 B6 2D 8C AB 57 DC 1C 02 35 AA A5 B3 31 00 00 00 00 00 00 00 BA 13 6E 91 2F
87 10 49 4E 46 50 A7 69 6A 19 53 E1 D6 14 53 30 6E 25 6D 00 C9 19 D7 DD B7 4A 08 21 D4 25 91 00 8C C5 B5 E3 40 9C 8F 15 E6 77 EA 6D EC B7 D3 FF D0 0E 8D AA 68 A8 BF D3 44 D5 30 3E 52 51 3D 7E 7B F7 27 71 16 7C ED AA 48 5E 1E 63 1A 6D 7A CE E8 4D 62 46 FA 44 AC EE A8 33 81 85 2B 7A 5D BE 93 9D B5 CF 2A 9F 2E B9 EC 88 A9 08 FA 9A 32 D2 F9 C3 C0 85 11 A9 8D 22 26 34 2F FD 7E 32 F1 07 2C 84 20 00 00 EC 75 E9 51 42 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 7A 38 BB D3 76 1D FE
AA 13 43 52 59 50 A7 69 6A 19 57 BE 26 C1 D9 89 5F EF E2 DF 78 41 A5 A2 12 0C 43 A6 F7 D2 41 36 34 6F A2 DC 05 62 A0 2B 12 E7 99 36 A1 42 98 D8 F5 22 C1 05 97 5F A0 18 1E F9 27 C9 53 D5 D7 AF B1 D2 78 6C 82 98 42 2B 77 77 45 38 0C 63 21 F1 5B C7 88 3B 9E 7D F0 C8 2E A3 38 E2 B2 F1 3A EA D9 4A 8B 15 B7 F6 58 61 BD D7 62 BD ED ED D2 D5 B8 E8 E4 59 7B 47 34 E3 5E 53 CB 3C 4B 21 4C E5 EC 4B 20 00 89 41 6E 5D C4 40 EC 0B 19 F9 27 06 FC 8F 90 3F 7A 17 78 5C 23 98 C3 64 AA E7 F5 26 EC 5F 04 DE 00 00 00 00 00 00 00 33 4B 1F D0 25
87 10 49 4E 46 50 A7 69 6A 19 DF 85 93 6C D9 AA 22 EC 6D 00 BE 31 B7 70 3F 0A 08 72 64 3A 36 94 CA F4 12 B5 03 B8 A9 AA 64 83 23 95 1F C2 F6 AB 90 B5 BA F6 5C 78 13 1B 59 ED 87 85 BD FD 43 6D 3F 73 09 A4 43 88 3B 4F 9E 9F FB 68 04 A8 E4 90 8C 81 B4 FF 95 B2 77 E1 B2 81 15 CE 6F 60 22 20 B9 DF 7F 52 AF 48 B8 A6 C6 12 00 66 18 1B DE 43 2E 5E 89 2D 98 9F E2 B7 CC 5A 42 80 D3 61 00 53 E0 4B 20 00 89 41 6E 5D 1B D0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 7A 38 BB D3 BE 28 6C
87 10 49 4E 46 50 A7 69 6A 19 BB 52 18 0E 95 ED F8 05 6D 00 19 CF FD E8 2D EA F9 80 6C 6F 5D B8 6C 21 00 B5 42 28 43 97 9A 80 70 E1 FF 77 D1 D0 B8 06 5C 47 1E 75 A1 1B 9C 8D ED EB 07 15 50 3A AE 4D F2 B4 D8 F1 32 C6 8C E1 6E 83 24 16 BA D2 CA 1A 03 AA 94 97 CA 32 D8 4A 46 E9 31 1F 63 0A B4 A3 05 F2 B4 13 42 D4 E1 6C 7F D5 9B BA 2E 1F 48 91 5C 82 C9 80 33 00 0D CE A1 19 13 1B 6B FD 90 65 20 00 9E 73 6E 8A 4A B9 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 7A 38 BB D3 32 C3 5B
87 10 49 4E 46 50 A7 69 6A 19 13 3D B3 E9 7A 08 E7 DC 6D 00 B5 73 8E 0C EC FF 50 D4 25 79 DD 68 8E E9 C2 C8 D6 4D 32 C8 01 05 0E F9 BC 14 CC DA 51 7D 91 E7 9A 10 82 B1 92 6E A2 7F E6 40 03 06 57 49 6A 30 26 B1 0A 7D 4B 9E 03 FE 7B F1 3C BC 46 9D C5 D1 18 A1 15 4E 5C E4 6F 12 55 F6 C3 72 F0 C3 3C 99 69 30 07 A9 31 16 B0 E1 5A AF E7 8E FF B2 8C AF B8 84 21 54 CF 7B 02 F6 77 80 81 85 8E 3B 20 00 21 EB 67 06 D7 D2 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 7A 38 BB D3 EF EB F1
A7 11 49 4E 46 50 A7 69 6A 19 4B 0A 04 27 9F BA BF 2D 8D 00 6F 5D BD 73 75 DD E4 20 F7 85 46 F1 69 3A F8 FA FB C2 15 E3 B6 CE 77 FC 46 91 2B F6 85 D3 B2 80 90 D0 52 DD A3 B4 C1 1C 4C 6D 1F 39 71 D7 52 F3 DD 17 31 DD 87 44 67 C1 0D 8B 09 62 36 05 91 78 54 1F 3B FA 2F F9 41 3F 22 92 3A F6 50 51 B4 2D 09 CF AF 04 E3 A6 2F D8 C9 11 D9 73 04 F1 24 D5 C7 6A 9F 05 A6 DF BD 07 86 FE 33 73 0D D6 33 17 C6 D7 EB 9D 80 DF B8 E1 8A B0 D3 07 E4 4C FE 43 34 56 6F 90 B2 2A 41 E4 B1 76 88 6A 33 F2 FA 5B 00 00 00 00 B1 DE 00 00 00 F2 5C 6F
AA 13 43 52 59 50 A7 69 6A 19 57 BE 26 C1 D9 89 5F EF E2 DF 78 41 A5 A2 12 0C 43 A6 F7 D2 41 36 34 6F A2 DC 05 62 A0 2B 12 E7 99 36 A1 42 98 D8 F5 22 C1 05 97 5F A0 18 1E F9 27 C9 53 D5 D7 AF B1 D2 78 6C 82 98 42 2B 77 77 45 38 0C 63 21 F1 5B C7 88 3B 9E 7D F0 C8 2E A3 38 E2 B2 F1 3A EA D9 4A 8B 15 B7 F6 58 61 BD D7 62 BD ED ED D2 D5 B8 E8 E4 59 7B 47 34 E3 5E 53 83 D6 55 66 C4 6C 97 7E 20 00 2A 2D F0 AE 79 0A 6A FD CC 07 29 B6 86 A9 7C CA AE 17 84 EC D8 47 B3 12 58 5F F3 CF 07 74 57 60 00 00 00 00 00 00 00 57 E8 8B F3 7A
87 10 49 4E 46 50 A7 69 6A 19 47 F6 D4 66 1A 66 AB CC 6D 00 78 A7 DF 61 3A 7E 06 B8 DA 77 BD 20 5B E2 B5 FB 69 94 02 4B 6A 3D FB F6 B8 CA 83 58 AA 7D D6 05 DF 18 36 AE 3B FF AB F3 0D F5 20 AE DF 62 A5 4A 5F 8F DF F5 17 68 74 8D 43 1A 4C D5 1C [... cut ...] 56 81 06 9F E2 CB 27 E8 62 57 7C 21 F8 48 A0 E5 E1 81 B8 86 CF BF 1E B8 3B 19 CA CA 14 BC 0C 67 52 6E 77 5C EC C3 18 34 0F 33 5D A4 77 64 83 32 52 04 5E 5F CB F4 90 64 05 2E 20 00 80 43 8C 0E 2E E3 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 7A 38 BB D3 8B 75 54
87 10 49 4E 46 50 A7 69 6A 19 FB 67 AE 8E 07 74 61 2A 6D 00 BD B6 96 E4 E9 91 D7 A4 EE 72 7A D0 E9 6A 07 5F A0 5C 75 B9 3B 44 62 B3 76 CB 55 7A 32 94 62 46 2D CE 42 F1 EA BB F0 62 0D BD 22

luyii-code-1, 2026-09-28:
Simply put, I currently have the ability to recognize packages, but not yet the ability to parse them. To put it bluntly, this is still a trade secret

june9713, 2026-10-03:
I own almost all NI USRP SDRs. If you'd like, I can record data using x401 or PXI equipment. I hope to participate in this project.