proto17/dji_droneid #51: ocysync 3

mutchup, 2024-06-09:
i have tried the same method on the ocusync 3 frame id and i get the frames but when i try to decode it fails have u ever tried it ?

mscanm, 2024-11-09:
Actually I also tried on ocusync 3 and I got turbo decoding and CRC check, I also got the frames
[80 10 49 4e 46 50 ac c4 24 0d 1a 5d 33 ec f2 6f 03 b8 6d 00 a1 cb 11 31 6a e8 b1 38 e4 9c 4e 30 4d 5c 88 57 11 0b 42 7d 21 73 1d 74 f4 7f 3f f7 b0 cc 8f f5 4e cb cc bd d4 68 0e a9 e3 30 91 49 ...]

PacketType 0x1001 or 0x1002 is not matched

(I also implemented for mavic 2 and it worked propoerly.)

mutchup, 2024-11-09:
I got the same results a while ago go but now I got some error in the turbo decoder when I use the octave any idea how to fix it ?

mscanm, 2024-11-09:
Actually I use the Turbo decoder from a cpp build. It is quite complex to use it octave or matlab

mutchup, 2024-11-09:
Same one as in the proto repo ?

mscanm, 2024-11-09:
Yes

mutchup, 2024-11-09:
That's weird mine just stopped working and I get the turbo decoder not found when it is in the directory and it has been built :_

jupercn, 2025-04-23:
i got the similar frames [80 10 49 4e 46 50 ...]，may be it is ocysync4.0.  ocusync3.0 drone after firmware update will update to ocusync4.0， it's encrypted,  can anyone known how to decrypt it?

palm578, 2025-08-21:
I also got the same result [80 10 49 4e 46...], with the crc check passed. And the signal is from DJI mini4 Pro, which is said to use Ocusync 4.0.