proto17/dji_droneid #53: ocusync 3.0 data packet structure

ozymandiasleo, 2024-11-19:
The OcuSync 3.0 packet structure does not match the structure described on this GitHub page and in the related articles. The serial number, drone location, and drone speed are not found anywhere in the data packet. Moreover, all the incoming drone ID packets are different, except for the first 10 bytes which are the same. Is there anyone who has successfully analyzed the OcuSync 3.0 data packet structure ?

mscanm, 2024-12-11:
Can your drone have a Government Edition version of Ocusync 3?

glianghan, 2025-08-21:
I'm using mavic2, but it shows a CRC error. Have you ever encountered this?

glianghan, 2025-08-21:
> > Actually I also tried on ocusync 3 and I got turbo decoding and CRC check, I also got the frames [80 10 49 4e 46 50 ac c4 24 0d 1a 5d 33 ec f2 6f 03 b8 6d 00 a1 cb 11 31 6a e8 b1 38 e4 9c 4e 30 4d 5c 88 57 11 0b 42 7d 21 73 1d 74 f4 7f 3f f7 b0 cc 8f f5 4e cb cc bd d4 68 0e a9 e3 30 91 49 ...]实际上，我还尝试了光同步 3, 得到了涡轮解码和 CRC 检查，我还得到了帧 [80 10 49 4e 46 50 ac c4 24 0d 1a 5d 33 ec f2 6f 03 b8 6d 00 a1 cb 11 31 6a e8 b1 38 e4 9c 4e 30 4d 5c 88 57 11 0b 42 7d 21 73 1d 74 f4 7f 3f 7f 3f f7 b0 cc 8f f5 4e cb cc bd d4 68 0e a9 e3 30 91 49...]。
> > PacketType 0x1001 or 0x1002 is not matchedPacketType 0x1001 或 0x1002 不匹配
> > (I also implemented for mavic 2 and it worked propoerly.)(我也为 mavic 2 实现了它，效果很好。)

I'm using mavic2, but it shows a CRC error. Have you ever encountered this?