proto17/dji_droneid #58: turbo decode matlab version (@ocusync3.0)

chiboluo98, 2025-03-14:
Hello! I recently tried to follow your work to extract the Drone ID based on OcuSync 3.0. During this process, I encountered the following questions and would like to seek your advice:

In the observed frame structure of the "suspected Drone ID" for OcuSync 3.0, I found ​10 OFDM symbols (as shown in the attached figure) ​without a long CP sequence. There are ​four ZC sequences. The number of subcarriers is consistent, and the modulation is QPSK. Ultimately, I obtained ​7200 bits of data.
[image]
[image]

Following your method, I performed descrambling. However, during ​Turbo decoding, since I am unfamiliar with C++, I opted to use MATLAB’s built-in functions. The code is as follows:

matlab
recovered_soft_bits = lteRateRecoverTurbo(turbo_decoder_input, 1408, 0);  
decoded_bits_o = lteTurboDecode(recovered_soft_bits, turbo_iterations);  
decoded_bits = decoded_bits_o{1};  
decoded_bits = decoded_bits(9:end-24);  
Here, turbo_decoder_input is the soft-valued sequence (size 1×7200), recovered_soft_bits is the rate-recovered sequence (size 1×4332), and turbo_iterations is set to 4. The final output decoded_bits_o is 1×1440. Notably, ​no CRC check was applied to my data.

I validated the encoding/decoding process experimentally, and it appears reasonable. The verification code is:

matlab
trBlkLen = 1408;  

codewordLen = 7200;  
rv = 0;  
crcPoly = '24A';  
tx = randi([0,1], 1, trBlkLen);  
trblockwithcrc = lteCRCEncode(tx, crcPoly);  
codeblocks = lteCodeBlockSegment(trblockwithcrc);  
turbocodedblocks = lteTurboEncode(codeblocks);  
codeword = lteRateMatchTurbo(turbocodedblocks, codewordLen, rv);  
rateRecovered = lteRateRecoverTurbo(codeword, trBlkLen, rv);  

decoded_bits_test = lteTurboDecode(rateRecovered, 4);  
decoded_bits_test_bit = decoded_bits_test{1};  
[blk1, err1] = lteCRCDecode(decoded_bits_test_bit(9:end), '24A');  
sum(abs(double(blk1)' - tx))  
**​Problem: Since I currently lack an OcuSync 2.0 device, I cannot confirm whether the Drone ID packet I identified is incorrect or if my Turbo decoding implementation is faulty. ​Could you kindly verify whether this MATLAB code fully simulates the corresponding C++ program?**

Thank you for your time and guidance!

chiboluo98, 2025-03-17:
> Does 7200 bits of data refer to 1200 used subcarriers times 6 OFDM symbols?
No. There are only 600 used subcarriers (without DC), but they're modulated by QPSK.
7200 bits = 600 subcarriers × 6 symbols × 2 bits per symbol (QPSK)

XDJQZ, 2025-03-17:
For the 1024 point NFFT, there are no obvious peaks in both locally generated ZC sequences and segmented cross-correlation. Could you please take a look at what the problem is

figure(1)
spectrogram(bf_ofdm,1e2,1e1,'yaxis','centered');

for i = 0:S_CSym
    R_CSym1(i+1) = sum( ZCseq1_upsample .* conj( bf_ofdm((1:N_upsamp)+i) ) );
end
figure(2)
stem(abs(R_CSym1));
index1 = find(abs(R_CSym1) == max(abs(R_CSym1)));

len = round(Nfft/2*fs/B);
R_CSym2 = zeros(1,S_CSym+1);
for i = 0:S_CSym
    R_CSym2(1,i+1) = sum( bf_ofdm((1:len)+i) .* conj( fliplr(bf_ofdm((1:len)+i+len)) ) );
end
figure(3)
stem(abs(R_CSym2));

chiboluo98, 2025-03-20:
How do you generate 'ZCseq1_upsample'? It appears that an incorrect ZC root may have been selected.
By the way, could you specify the UAV model or the communication protocol that was used to acquire the raw data? I do believe that we have identified a 'suspected drone ID' exhibiting the same frame structure.

XDJQZ, 2025-03-20:
> 2 solutions you may try: 1.reverse IQ data 2.don't resample

Neither of these methods works, please leave your contact information if possible, and I can send you non private IQ data

m1nl, 2025-03-20:
this packet is not drone id, it's sent by drone when it's stationary and not connected to RC; ZC seqs are 186, 186, 600, 600 (symbols in the middle); CPs are 72 for all 10 symbols; I've also received it but I'm unable to extract payload, I believe different scrambling sequence is used

chiboluo98, 2025-03-21:
> this packet is not drone id, it's sent by drone when it's stationary and not connected to RC; ZC seqs are 186, 186, 600, 600 (symbols in the middle); CPs are 72 for all 10 symbols; I've also received it but I'm unable to extract payload, I believe different scrambling sequence is used

Thank you! I was also curious whether this packet is associated with a drone ID since its duration is only about 200 ms—and I captured it indoors without any drone in flight.  And the length of  all cps are 72 symbols. However, the ZC sequence root I obtained is 15, 15, 600, 600.

chiboluo98, 2025-03-21:
> Neither of these methods works, please leave your contact information if possible, and I can send you non private IQ data

I suggest estimating the frequency offset and verifying the local ZC sequence. The author provided MATLAB code to generate the ZC sequence, which should work since the packet spectrogram you attached clearly shows ZC sequence characteristics.

XDJQZ, 2025-03-21:
> > 此数据包不是无人机 ID，它是无人机在静止且未连接到 RC 时发送的；ZC 序列为 186、186、600、600（中间的符号）；所有 10 个符号的 CP 均为 72；我也收到了它，但我无法提取有效载荷，我相信使用了不同的扰码序列
> 
> 谢谢！我还很好奇这个数据包是否与无人机 ID 相关联，因为它的持续时间只有大约 200 毫秒——而且我是在室内捕获的，没有任何无人机在飞行。所有 cps 的长度都是 72 个符号。然而，我获得的 ZC 序列根是 15、15、600、600。

Isn't the duration of this signal frame about 700ms？

XDJQZ, 2025-03-21:
> > 这两种方法都行不通，如果可以的话请留下你的联系方式，我可以给你发送非私密的IQ数据
> 
> 我建议估算频率偏移并验证局部 ZC 序列。作者提供了 MATLAB 代码来生成 ZC 序列，这应该可以工作，因为您附加的数据包频谱图清楚地显示了 ZC 序列特征。

Thank you, I will try this method

alphafox02, 2025-03-21:
I’ve not been following this totally, but with the AntSDR and its droneid implementation onboard, today it decoded properly a Mavic 3. Isn’t that ocusync 3?

XDJQZ, 2025-03-21:
> 我并没有完全关注这个，但是通过 AntSDR 及其 droneid 实现，今天它正确解码了 Mavic 3。这不是 ocusync 3 吗？

Okay, I will get back to you after further simulation attempts

XDJQZ, 2025-03-23:
> I suggest estimating the frequency offset and verifying the local ZC sequence. The author provided MATLAB code to generate the ZC sequence, which should work since the packet spectrogram you [...]

Despite frequency offset estimation and compensation, there is still no peak in the ZC sequence correlation results, and the constellation diagram is not correct.

hechunzhe, 2025-03-24:
Hello, I collected the signal from the DJI Mavic 3 drone today and observed the same frame structure as you. Could you please confirm if it is a DroneID frame?

chiboluo98, 2025-03-26:
> Hello, I collected the signal from the DJI Mavic 3 drone today and observed the same frame structure as you. Could you please confirm if it is a DroneID frame?

It's not DroneID frame as far as i know.

hechunzhe, 2025-03-26:
> > Hello, I collected the signal from the DJI Mavic 3 drone today and observed the same frame structure as you. Could you please confirm if it is a DroneID frame?
> 
> It's not DroneID frame as far as i know.

Thank you for your reply. Have you collected any DroneID frames afterwards? Do you need to have the drone take off while capturing the signal?

chiboluo98, 2025-03-28:
> Thank you for your reply. Have you collected any DroneID frames afterwards? Do you need to have the drone take off while capturing the signal?

I believe the drone ID is only transmitted after takeoff. I’ve tried many times indoors without flying the drone and didn’t detect any packets resembling the drone ID. However, since drones are not allowed to fly in my city, I haven’t had the chance to test this outdoors yet. The drone I own uses ​Ocusync 3.0 Enterprise protocol, which is encrypted with ​AES-256 as claimed by DJI. I'm not sure I can decode the Drone ID afterwards.

m1nl, 2025-03-28:
I have collected DroneID from ocusync 3.0; same frame format as before, but encrypted.

hechunzhe, 2025-03-31:
> I have collected DroneID from ocusync 3.0; same frame format as before, but encrypted.

How did you collect the data? Is the frame format of the DroneID in OcuSync 3.0 the same as the frame format in OcuSync 3.0? Looking forward to your reply.

chiboluo98, 2025-03-31:
> I have collected DroneID from ocusync 3.0; same frame format as before, but encrypted.

How did you determine that it’s encrypted? Is it because the CRC check passes but the data can’t be parsed? I’m wondering if they might be using a different scrambling code. I’ve been thinking—if it’s encrypted, how does DJI’s Aeroscope software obtain the Drone ID?

m1nl, 2025-04-02:
frame format is the same, in my case first symbol is missing (the one which is zeroed when descrambled); CRC check passes and I'm unable to decode the content; checks for entropy suggest the content is encrypted

metusa, 2025-06-03:
I got 176 byte frame for mavic 3. The frame ends with .....ffffffffffffffd8c8bd. 
7 byte ff + 3 byte CRC. Did anyone decode the content ? Any idea about which parts are encrypted, encryption type?

King-Of-Knights, 2025-08-24:
@m1nl O3 

> I have collected DroneID from ocusync 3.0; same frame format as before, but encrypted.

O3 is not encrypted, but O3+ does

[image]

lj411, 2026-02-18:
> 您好！我最近尝试按照您的方法，基于 OcuSync 3.0 提取无人机 ID。在此过程中，我遇到了以下问题，希望得到您的解答：
> 
> 在OcuSync 3.0的“疑似无人机ID”观测帧结构中，我发现了10个OFDM符号（如图所示），这些符号没有较长的循环前缀（CP）序列。此外，还有四个零序（ZC）序列。子载波数量一致，调制方式为QPSK。最终，我获得了7200比特的数据。 [image] [image]
> 
> 
Thank you for your previous help with decoding RID frames. Recently, I found that RID frames and these so-called suspected search frames can appear simultaneously; this phenomenon has been observed on both the DJI MINI 4 PRO and the DJI AVATA 2. If the search frames are transmitted when the remote controller and the drone are not connected, how can the drone simultaneously broadcast RID frames? This does not seem logical, and it also raises the question of how the pilot-related information is determined during frame construction.

[image]