proto17/dji_droneid #60: Struggling in Signal Recording

mr-aq-stack, 2025-05-23:
Hi Sir,

I hope you're doing well. I need your assistance with recording the drone signal.

I'm currently using your GNU Radio flowgraph along with the recommended parameters (e.g., sample rate of 30.72 MSPS) to record the signal. After capturing, I inspected the recording in inspectrum, and you can see the result here:

Image

However, when I try to decode the recorded file using the process_file.m script, it consistently results in CRC errors.

Could you please advise on how you recorded the drone signal? Any brief guidance or confirmation of the correct setup would be greatly appreciated.

Best regards,

[image].

mr-aq-stack, 2025-08-22:
Thanks for the clarification. Could you share which hardware (sdr etc) you are using and what your RX configuration looks like for this setup?

mr-aq-stack, 2025-08-22:
That’s really interesting — thanks a lot for sharing the details .The spectrum pic is super helpful too — really appreciate you!

If you have a chance, would you be open to trying to decode these packets ? I want to see what kind of frames pop out. And if you’re okay with it, kindly share one of your IQ recordings at axen.qadeer@gmail.com ? That would help a lot for testing different decoding algorithms.

King-Of-Knights, 2025-08-22:
@mr-aq-stack I have sent you with O2, O3, O3+, O4 sample. O2, O3 can be parsed, O3+, O4 are encrypted.

mr-aq-stack, 2025-08-22:
Thanks a lot. I'll check their spectrum and let you know.

mr-aq-stack, 2025-08-22:
Yes there is droneid signal in your IQ file. Here is its spectrum:

[image]

Do you have decoded it ? or is there any approach to decode it ?

djiusermavic04, 2025-08-22:
Hey @King-Of-Knights   I have done O and O2,  if you don't mind can you share to me to this email  djiusermavic04@gmail.com  your IQ recordings  O2, O3, O3 , O4 sample, thank you an adavnce, also where can I get the SDR and ANttena that you have been using

hopeluke0501, 2025-09-01:
Hi, @King-Of-Knights, I'm currently trying to decode the drone ID, but I've encountered some difficulties (I can't record  the expected drone ID signal, the sgnal I received  seems to be the video signal ). Could you share the receiver parameters you used, such as the sampling rate and receiving frequency? Additionally, could you share the drone ID IQ sample  you collected for DJI O2, O3, O3+, and O4? hopeluke@163.com
Thanks a lot.

> Hey [@King-Of-Knights](https://github.com/King-Of-Knights) I have done O and O2, if you don't mind can you share to me to this email [djiusermavic04@gmail.com](mailto:djiusermavic04@gmail.com) your IQ recordings O2, O3, O3 , O4 sample, thank you an adavnce, also where can I get the SDR and ANttena that you have been using

alphafox02, 2025-09-01:
> > Thanks for the clarification. Could you share which hardware (sdr etc) you are using and what your RX configuration looks like for this setup?
> 
> I am using [Y380](https://v3best.com/y380) with a 122.88 MHz sample rate and 100 MHz bandwidth. It looks like DroneID uses a frequency-hopping strategy and hard to catch (it can even jump from 5.8GHz to 2.4GHz 😂). From my experience, you may need a wideband receiver to observe all 2.4GHz and 5.8GHz channel so you won't miss it. Collecting some IQ data and using [signal-analyzer](https://ww2.mathworks.cn/help/signal/signal-analyzer.html) to find out the potential one. Also, I have a pic for you, you can see DroneID (left-bottom) and image transmission signal (Yes, it seems also used ZC sequence)
> 
> <img alt="Image" width="1797" height="1162" src="https://private-user-images.githubusercontent.com/25562792/480907313-af047c48-fc14-44dc-8e46-5c0a575cb3aa.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTY2OTczOTksIm5iZiI6MTc1NjY5NzA5OSwicGF0aCI6Ii8yNTU2Mjc5Mi80ODA5MDczMTMtYWYwNDdjNDgtZmMxNC00NGRjLThlNDYtNWMwYTU3NWNiM2FhLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTA5MDElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwOTAxVDAzMjQ1OVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWZlZmU4ODNkZGYxZjM1NjMyZTBhMWJkYzkzZGI2NzQ2ZGM4MTBiZWZiNTZlNTNiNjVjZGQ1OTlhZDE4ZDVlMjAmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.UlcIHFz60RosAvFQeINETT0g1N7PsPc2hba9aztwbc0">

Little off topic, but how could I get one of those y380’s to play with? I can’t really tell if it’s possible to order one in the US or how much it’d cost, but pretty cool looking.

King-Of-Knights, 2025-10-20:
@pianeer0 发了

yasin112358, 2025-11-26:
Hi , can you send O1 ,O2, O3 samples to me please. yasinozbunar@gmail.com

King-Of-Knights, 2026-02-09:
> Hi everyone,
> 
> Apologies for jumping into the conversation, but I'm trying to decode the drone ID of a DJI Mavic 3 Classic (which should use Ocusync 3.0+), and I’m stuck on the signal demodulation.
> 
> The script finds the packet correctly, but fails to demodulate it. When I try to print the signal constellation, I get a donut pattern like the one in the image below.
> 
> <img alt="Image" width="800" height="800" src="https://private-user-images.githubusercontent.com/64355437/547241724-611100bf-ecb1-4bca-af84-c9ecbd691cd2.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NzA2NjQzMzQsIm5iZiI6MTc3MDY2NDAzNCwicGF0aCI6Ii82NDM1NTQzNy81NDcyNDE3MjQtNjExMTAwYmYtZWNiMS00YmNhLWFmODQtYzllY2JkNjkxY2QyLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNjAyMDklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjYwMjA5VDE5MDcxNFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTRmZmYyMTkwZmNiMGU2ZGEwYzkyZGQwNTA2ZWQxODIzZTZkNzk1ZTJiZTYyOWQ2OWNjZjZmOTBhYzJiNWQ1ZTQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0._mw2f7qFd-QGyrTPsqbICJZT4QUAPRU3UjCsp_g97fA">
> Do you have any suggestions on how to proceed? Thank you for your time.

it seem that your script didnt find the correct Signal start point or carrier offset

Clauze, 2026-02-12:
> > Hi everyone,
> > Apologies for jumping into the conversation, but I'm trying to decode the drone ID of a DJI Mavic 3 Classic (which should use Ocusync 3.0+), and I’m stuck on the signal demodulation.
> > The script finds the packet correctly, but fails to demodulate it. When I try to print the signal constellation, I get a donut pattern like the one in the image below.
> > <img alt="Image" width="800" height="800" src="https://private-user-images.githubusercontent.com/64355437/547241724-611100bf-ecb1-4bca-af84-c9ecbd691cd2.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NzA4OTA5MzgsIm5iZiI6MTc3MDg5MDYzOCwicGF0aCI6Ii82NDM1NTQzNy81NDcyNDE3MjQtNjExMTAwYmYtZWNiMS00YmNhLWFmODQtYzllY2JkNjkxY2QyLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNjAyMTIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjYwMjEyVDEwMDM1OFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTQ1MmMyY2E2Mjg0ZWUwZGQ0OGQzN2Q3YmQyNWIzYmE3OGI3YjFhOTIxNjA5NTkyNmE3Y2Q0NDQ2M2M1MGIwYzkmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.UlC5sID-vlxzdcIcylGucDjskpebOqYVa63QV-Zf_oI">
> > Do you have any suggestions on how to proceed? Thank you for your time.
> 
> I believe 3+ and 4+ are end to end encrypted so whilst detection is possible, decoding is not.

So you mean there is some sort of physical layer encryption?

King-Of-Knights, 2026-02-12:
> (80 for the first and 72 for the rest)

80 for first and last one, remain for 72

Clauze, 2026-02-21:
Sorry to bother you again! I believe I've resolved my previous issue. Currently, when I run the code, the constellation diagram looks good as you can see in the image, however the decoded payload for my drone id packet for DJI Mavic 3 Classic consistently returns long strings of ff:

FRAME: 00adb7ffffffffffffffffffffffffffffffffffffffffffffffffffff...7709d6

I suspect a descrambling issue or a synchronization offset. I noticed my packets have 8 symbols while the script expects 9. I tried modifying the processing logic to handle 8 symbols, but the output and the plots remain exactly the same, which makes me think I might be missing a hardcoded parameter elsewhere.

Do you have any suggestion for this problem? Do someone of you also face it?
I'm seeing this same 0xFF pattern across all captured packets.

Thanks for your time!

[image]

King-Of-Knights, 2026-02-25:
> > [@Clauze](https://github.com/Clauze) You could send me your data, I will return my result to help you check. 8 symbol droneid is enough
> 
> I uploaded some recordings that I used in MATLAB to this link. I resampled them at 15.36 MHz: https://drive.google.com/drive/folders/1ac92K11h-X2Pjxj6kzYwaM76cfhXGIva?usp=sharing Also, I added one recording called droneid_record_40Mhz that I extracted from the original recording at 40 MHz (so without resampling it to 15.36 MHz).
> 
> Thank you for your time, I'm completely stuck😔.

Hi, you need to upload a matlab script to tell me how to read your sample file. If you could directly send me a mat file that contains complex-valued IQ waveform, that will be more easier

King-Of-Knights, 2026-02-25:
> > > > > I'm using a Mavic 3 Classic and an USRP B200. The recording was done indoors at my University in Padua because we weren't allowed to go outside. Could the lack of a GPS lock be causing the string of FF values?
> > > > > Or could there be something different than Ocusync 2.0 in the descambing and turbo decoding phase?
> > > > 
> > > > 
> > > > No, nothing change in descambing and turbo decoding phase from O2 to O4+. Have you link your drone to your controller? Drone can get GPS information from controller
> > > 
> > > 
> > > Yes, from the controller I could see the drone's live video recording.
> > 
> > 
> > That's wired, can you find Flight control SN from the controller?
> 
> Unfortunately, I don't know this and can't verify it because I no longer have the drone. So, the controller sends the position to the drone, which then creates the drone-id packet and redistributes it. Could it receive a corrupted position and create this garbage packet? But shouldn't I still be able to get some basic data, like the packet length or the start bytes?

There is no need to investigate it anymore, the good new is you have a complete demodulation link and try another Drone

Clauze, 2026-02-25:
> There is no need to investigate it anymore, the good new is you have a complete demodulation link and try another Drone

Great, thanks for this information. I would have liked to finish and get a complete, properly decoded package. Do you have any idea what might be causing these problems? Most importantly, I don't want to make the same mistake on other recordings.

Clauze, 2026-02-25:
> I've never encountered this before. Perhaps the system has been modified?

Do you mean the firmware? 
I don't think so. The drone is from the university, and I don't think the professor modified it. 
There might be some small differences in the package if the drone was purchased on different continents, such as the European or Chinese version, but I don't think I should see these anyway.

LucasCombe-flyingeye, 2026-02-27:
Hi @Clauze , I am struggling at the same place as you were 3 weeks ago : I recorded DroneID in O4 using DJI Mini 4 Pro (I know you were using O3 but this shouldn't change much). The script finds finds the packet correctly, but fails to demodulate it and I get donut patterns just like you. How did you manage to  fix this issue ? I'm recording with a URSP B210 at 50Mbps and I tried resampling to 30.72MHz and 15.36MHz and the issue stays the same.
Thanks in advance for your answer !

Clauze, 2026-02-27:
> Hi [@Clauze](https://github.com/Clauze) , I am struggling at the same place as you were 3 weeks ago : I recorded DroneID in O4 using DJI Mini 4 Pro (I know you were using O3 but this shouldn't [...]

Hi!

Actually, I was able to solve the problem using another recorded Drone ID packet. Moreover, the previous plot where I had the error comes from another repo (https://github.com/RUB-SysSec/DroneSecurity) with some adjustments. Unfortunately, I don’t have any other suggestions for your problem; however, I found this issue that may be useful since the plot appears similar to mine and I believe yours as well: https://github.com/proto17/dji_droneid/issues/25.

I hope this helps!

TheOldCode, 2026-03-05:
> [@waseem-arshad94](https://github.com/waseem-arshad94) Sorry for the late reply, actually I am also struggling in recording DroneID automatically. Nowaday, I just manually slice the DroneID from a long IQ seqence (look and find manually ).

@King-Of-Knights ,could you share the drone ID IQ sample you collected for DJI O2, O3, O3+, and O4? (songzhenfeng67@gmail.com) really appreciate you!

pandajr69, 2026-07-17:
@King-Of-Knights  I want to ask you again, DJI drone id center freuquencies are fixed? or are they random frequencies ?

And, my other question is, can we use that repo for other drones except dji ? (autel etc.)

pandajr69, 2026-08-28:
@frbarcio 

I’m not having any trouble detecting bursts—it tells me when it finds one—but the problem is that the bursts are false positives. In other words, it detects a burst but can’t pass the CRC check. I’m not sure what I should do about this. I’m currently using an SDR with a 92.16 sampling rate and resampling it to 15.36 to test it. But I don’t think the problem is caused by the SDR because the same issue occurred with Pluto and USRP as well. 

I can’t set up an FPGA + filterbank right now, but since that’s a guaranteed solution, I need to switch to it. Still, I want to be sure the code is correct. 

The 2.4 GHz band is already noisy (Wi-Fi + Bluetooth), while 5.8 GHz is cleaner. Still, since it’s known to hop randomly, there’s a chance I won’t be able to capture it no matter how long I wait. 

I’m wondering about this: Have you made any changes or improvements to the current code? Is there even the slightest possibility of an error in the `remove_crc` step? I say “slight” because it’s not that it never happens—it just doesn’t happen often.

pandajr69, 2026-09-10:
i catch sometimes different packet, it can pass the crc and give me serial number. but i can not decode.

what is that packet? do  @King-Of-Knights @frbarcio know that packet how could i parse ? 

[14:37:06] 2414.5 MHz  EVM=43.2%  CRC OK
    type=0x11 len=141 serial=3YTBJ4D00301DX  data@: 18 19 20
  type=0x11 len=141 serial=3YTBJ4D00301DX  data@: 18 19 20
```
    0: 8d 11 33 59 54 42 4a 34 44 30 30 33 30 31 44 58
   16: 00 00 cc aa 02 00 00 00 00 00 00 00 00 00 00 00
   32: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
   48: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
   64: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
   80: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
   96: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  112: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  128: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 8c e7
  144: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
  160: ff ff ff ff ff ff ff ff ff ff ff ff ff 9c cb c9
```

pandajr69, 2026-09-11:
i catch sometimes different packet, it can pass the crc and give me serial number. but i can not decode.

what is that packet? do @King-Of-Knights @frbarcio know that packet how could i parse ?

[14:37:06] 2414.5 MHz EVM=43.2% CRC OK
type=0x11 len=141 serial=myserial data@: 18 19 20
type=0x11 len=141 serial=myserial data@: 18 19 2

```
    0: 8d 11 33 59 54 42 4a 34 44 30 30 33 30 31 44 58
   16: 00 00 cc aa 02 00 00 00 00 00 00 00 00 00 00 00
   32: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
   48: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
   64: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
   80: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
   96: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  112: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  128: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 8c e7
  144: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
  160: ff ff ff ff ff ff ff ff ff ff ff ff ff 9c cb c9
```

King-Of-Knights, 2026-09-11:
@pandajr69  Yes, I encounter this package very often and dont have idea about it neither. You could directly ignore it and focus on data package started with` 58 10`

pandajr69, 2026-09-11:
@King-Of-Knights  thank you

frbarcio, 2026-09-11:
@pandajr69 
Should be a "Licence packet"  There are three types of drone ID packets that the Paper "_DJI drone IDs are not encrypted_"
by Conner Bender has identified: 
license, (type 0x11)
flight information version 1 (type 0x1001) and
flight information version 2 (type 0x1002).

According to the paper: . ..After the packet type comes a serial number of the detected drone. Following the drone serial number are the custom license and flight plan values provided by the DJI GO smartphone application user...
(probalby only using the DJI application to program the flight plan the packet could carry something different from .00..s)

Here is an example from my captures:
(you can find here the CrC position "c449" - it matches)

```
Payload (176 bytes)
0000: 8D 11 31 36 33 43 47 41 34 52 30 41 33 30 54 48  |..163CGA4R0A30TH|
0010: 00 00 CC AA 02 05 00 0A 40 23 40 00 00 00 00 00  |........@#@.....|
0020: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  |................|
0030: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  |................|
0040: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  |................|
0050: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  |................|
0060: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  |................|
0070: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  |................|
0080: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 49 C4  |..............I.|
0090: 96 51 17 70 34 4F A9 19 82 B2 C3 B2 F0 A9 CE 96  |.Q.p4O..........|
00A0: 05 45 5D 2C 43 2C 59 62 EB C3 8C D2 36 0D D3 93  |.E],C,Yb....6...|
==============================================
```

## Drone-ID Payload ##
{
    "format": "license_packet",
    "pkt_len": 141,
    "packet_type": "0x11",
    "serial_number": "163CGA4R0A30TH",
    "crc-packet": "c449",
    "crc-calculated": "c449",
    "crc_ok": true
}

License packet: no coordinates

King-Of-Knights, 2026-09-11:
> [@pandajr69](https://github.com/pandajr69) Should be a "Licence packet" There are three types of drone ID packets that the Paper "_DJI drone IDs are not encrypted_" by Conner Bender has identified: [...]

Thanks, bro. That's a real mystery to me

King-Of-Knights, 2026-09-14:
@frbarcio @pandajr69 To make it more complete,

`packet_type` should be those below:
`0x10` data package
`0x11` licence package
`0x13` O3+ O4 key package

But I haven't met `0x12` before

pandajr69, 2026-09-15:
@King-Of-Knights @frbarcio 

I am wondering, I can set video signal band on DJI app (2.4 / 5.8 / Dual) and when I set any where for example I set 2.4, I can catch DJI drone id 2.4 or 5.8, It is jumping. Can I set drone id signal only one band like video ? Is there any way ? I think no way, but I would like to ask.

I am wondering about another think, did you tested when flying? how many distance meters, catch drone id signal?

pandajr69, 2026-09-15:
> [@frbarcio](https://github.com/frbarcio) [@pandajr69](https://github.com/pandajr69) To make it more complete,
> 
> `packet_type` should be those below: `0x10` data package `0x11` licence package `0x13` O3+ O4 key package
> 
> But I haven't met `0x12` before

I have air2s and I can not receive 0x13 packet, but If I receive any 0x13 packet from different drones, can I solve them or I need server for them (03+, 04)

King-Of-Knights, 2026-09-15:
> Can I set drone id signal only one band like video ? Is there any way ? I think no way, but I would like to ask.

There is no way to control it

> I have air2s and I can not receive 0x13 packet, but If I receive any 0x13 packet from different drones, can I solve them or I need server for them (03+, 04)

You need decrypt server. Besides, you need key package (started with `aa` or `a3`) and data package (started with `80` or `87`) from the same session (from first hex to fourth hex after `4e 46 50`, it will change when drone reboot) to decrypt. 

For example:
data package hex:
`8710494e465072832f8e01bfad5b9be4b4b96d003b3620d96914e13f34cb1a5dfce5990aba6efeec60a9ea264b28e8b4891808d38c3ac77b26d523126cf1b666f27e342a3f62d4045f02ec228cb27c2bbb6d486e86d6a7b7a9875da94cc9fa8b9f0fb9b4c04dc5f0a3afcc4835e642a20a0afe3922c2c42b598e3a9479b7d3a348182000e09d6c9c79e1000000000000000000000000000000000000000000000000000000000000007a38bbd3ec90da`

key package hex:
`aa134352595072832f8eb62e69864ad1c6c1d839e825d97339a19709f85b09769be69be1a3c3a80f8d4ff0d8c86a57e0d73e26d44e8d706972778ad70506e320729e87be661b443faab2ca533e44d94fae07696eb46d257e6fcc93761c24d9fbc4d657ea106bcf8c43be531c34ea74000adde8e1029990f308c2a510705686a6294b2000c7d91e28f28a4e397da0ea238c379e3528953d62b8385d53eda967093f2f9cd800000000000000286e6e2d12`

The SessionID should be `72832f8`. If you feed AeroScope with data package hex and key package hex, it will return the result below:

```
{
	"code": 0,
	"msg": "",
	"data": {
		"decryptType": "DATA",
		"keyDecrypt": null,
		"dataDecrypt": {
			"msg": null,
			"sn": "1581F986425C100ST1EL",
			"uuid": "1842775117804707840",
			"type": "108",
			"lon": 117.015136,
			"lat": 36.675705,
			"height": 36.5,
			"alt": 54,
			"x": -0.23,
			"y": 6.34,
			"z": 0,
			"yaw": 117.67,
			"gpsTime": "2026-08-31 17:30:28",
			"pilotLon": 117.014254,
			"pilotLat": 36.676942,
			"homeLon": 117.014242,
			"homeLat": 36.676896,
			"model": "Mavic 4 Pro"
		}
	}
}
```

Don't ask me where to buy AeroScope, someone just sent me API which will give the result from AeroScope.

Someone has claimed he has cracked it:

https://www.sdrfamily.com/product/drone-detector-and-locator

pandajr69, 2026-09-15:
@King-Of-Knights  

"You need decrypt server"

Yes I know, if use server, need network connection.

Does not that [sdr](https://www.sdrfamily.com/product/drone-detector-and-locator)  need server ?  If it does not need server, anyone can copy it. Am I right ?

King-Of-Knights, 2026-09-15:
> Does not that [sdr](https://www.sdrfamily.com/product/drone-detector-and-locator) need server ? If it does not need server, anyone can copy it. Am I right ?

I have been told they could decode O4 offline

pandajr69, 2026-09-16:
@King-Of-Knights 

In the that [video](https://www.youtube.com/watch?v=fYl9GiSUIIw), they said "Network Decoding". I don't know

[image]

pandajr69, 2026-09-22:
> [@King-Of-Knights](https://github.com/King-Of-Knights) Unfortunately not. I found some patterns that seem to repeat themselves, but there are irregularities, and they do not constitute a sequence. The issue is that I could not find any "slot" size, therefore what is the pattern, without a slot size ? If I were sure about the slot duration, I could try to sample and log a very long sequence to analyze.

@frbarcio could you share your founded pattern (_I found some patterns that seem to repeat themselves_) and your drone model?

King-Of-Knights, 2026-09-22:
@pandajr69 I think it may related to license package, I have observed that DroneID will hop to another frequency 8 package later when license package occur

pandajr69, 2026-09-22:
@King-Of-Knights  its not about license package, I wonder what he found, and If it is really true hopping list, I would like to try. he said that "_there are irregularities, and they do not constitute a sequence_" may be he lost some packages, and that list may be true and so for that, I might try if he share the list and drone model

frbarcio, 2026-09-25:
@pandajr69 
@King-Of-Knights
I found the pseudorandom sequence, at least for Mavic 2 under bench-test conditions.
Each slot consists of a train of 13 bursts (0,64 x13 = 8,32 secs).
The sequence comprises 32 slots (266,24 secs, almost 5 minutes).
Arranging the 8 channels by increasing frequency (0 to 7), the sequence I found is:
22217401 24625344 74630351 02657265

The channels are not used uniformly
The 2444.5 MHz frequency appears 7 times, while the others appear between 3 and 5 times.
This makes sense, considering that Wi-Fi channels 1 and 6 are often occupied and taking into account the location of the BLE advertising channels.
However, for the moment, this is the only information that is useful to me.

Achieving precise synchronization is not easy, as the bursts within the series do not indicate their position (as far as I know);
consequently, detecting a burst provides no information regarding the transition to the next slot.
In any case, I prefer wideband reception without retuning (retuning entails several disadvantages during decoding).
The sequence is quite slow; typically, the goal is to detect and react much more quickly.

In any case, I would be curious to know if your DJI drones exhibit the same sequence.
It is possible that the sequence varies not only between different models but perhaps also based on the mission's operating conditions.
(This might explain why, at my workstation, one channel appears to be used more than the others).

[image]

King-Of-Knights, 2026-10-03:
@pandajr69 

> In the that [video](https://www.youtube.com/watch?v=fYl9GiSUIIw), they said "Network Decoding". I don't know, probably we need network connection.

Lately, I've been focusing entirely on O4 offline decryption, and here are the [results](https://github.com/luyii-code-1/dji-ocusync-droneid-research/issues/1#issuecomment-5957785456) I've achieved so far. If you have a DJI Aeroscope Dongle, feel free to reach out so we can exchange ideas.

@frbarcio 

Apologies for the delayed response. I've recently been running continuous packet captures on a Mini 2 drone inside an RF anechoic chamber. After analyzing over 9,000 DroneIDs with consecutive sequence numbers, I can confirm that the transmission period is indeed 640ms. It transmits exactly 13 packets on a single frequency before hopping. The frequency hopping pattern is as follows, and it remains completely unaffected by drone reboots. Additionally, it appears that just before a frequency hop occurs, the drone retransmits data packets:

[image]

Perhaps we can derive a mathematical formula for this frequency hopping pattern, but I can't guarantee that other drone models follow the exact same rule.

frbarcio, 2026-10-05:
@King-Of-Knights

The sequence parameters (number of slots = 32, number of bursts = 13, duration = 640 msec) are consistent with my measurements.
Yet, the sequence is completely different, and I am quite surprised to see the occurrences:

[image]

One channel is never used!
(in the sequence I detected, for Mavic 2, the occurrences where 3 to 7, all 8 channels used).

Really strange...
Needs further investigations.

King-Of-Knights, 2026-10-05:
@frbarcio would you please share 2.4G frequency hopping sequence image? I will investigate other drone at my hand, I think it may related to product type

frbarcio, 2026-10-05:
@King-Of-Knights

You mean something like this?

[image]

Arranging the 8 channels by increasing frequency (0 to 7), the sequence I found is:
22217401 24625344 74630351 02657265

King-Of-Knights, 2026-10-05:
@frbarcio Yes, I will look into it

King-Of-Knights, 2026-10-05:
@frbarcio Hi frbarcio, do you have two drones which share the same product type but different SN. We need to exclude SN influence first

frbarcio, 2026-10-06:
@King-Of-Knights

Unfortunately not...

pandajr69, 2026-10-07:
@King-Of-Knights 

When I checked the frequency list you provided, mine doesn’t seem to match.

Actually, what I’m wondering is: Is the list the same for drones of the same model?

Could factors like different countries and different software versions be causing the discrepancy?

I think we should check these things.

1. Do drones of the same model—for example, two Mini 2 drones—have the same jump pattern?
2. If not, could the software version or different countries be affecting this?

frbarcio, 2026-10-07:
@King-Of-Knights
@pandajr69 

It is very palusible drones are adopting orthogonal pseudorandom sequences.
The asynchronous transmission with 0.1% duty cycle (burst length = nine consecutive LTE symbols = approximately 642.7–643.2 µs, versus  640 msec channel slot time) does not guarantee absence of collisions alone, two drones with overlapping burst timing could collide 13 times consecutively. Adding psudorandom channel selection decreases the probability of continuous collisions.
I would expect for sure different models use different sequence (easy at FW level), maybe also different SN ? (little more complicated, but feasible with Tables etc.).
Does not help much, anyway, whoever is trying to intercept. I assume Aeroscope has 8 full parallel receiving channels, each channel receiving all Beacons from all drones in the area.

King-Of-Knights, 2026-10-07:
> It is very palusible drones are adopting orthogonal pseudorandom sequences

It sounds interesting and make sense. DJI invent Ocusync by largely borrowing from LTE protocol

> I assume Aeroscope has 8 full parallel receiving channels, each channel receiving all Beacons from all drones in the area.

Yes, if you don't know the drone type and SN in advance, you have to monitor 8 channel simultaneously.

Also, aeroscope firmware is easy to analysis, we could verify your thinking easily

EdwardBlair, 2026-10-08:
The carrier sequence is not an orthogonal pseudorandom sequence. It is driven by an incrementing 8-bit publication counter and a fixed **seven-lane** lookup.

The counter wraps after 256 publications. At 640 ms per publication, the complete supercycle is exactly 163.84 seconds. The modulo-256 wrap must be applied before the modulo-seven lookup, so it is not simply an endlessly repeating seven-channel loop.

I’ll publish my receive-only follower code. It acquires integrity-valid CRYP/INFP objects, retains the possible counter phases, and follows the schedule once the phase becomes unique. Schedule phase alone does not identify an aircraft; concurrent sessions must remain separated using validated session fields from the packet.

Observations from older drones (your Mini 2/Mavic 2) should not be generalized to O4.

Following becomes challenging with multiple drones, since a crowded airspace requires a clever scheduler to track independent packet phases and allocate limited receiver time across competing sessions. With sufficient instantaneous bandwidth, wideband capture and channelization are simpler. This is what aeroscope does. It has 8 SDR inside it.

frbarcio, 2026-10-08:
@EdwardBlair 
Thanks for explanation. Yet, I measure 32 x 13 = 416 burst periods (your "publication"), and 266,24 seconds, before the supercycle repeats.
Maybe this applies to old drones only ?

EdwardBlair, 2026-10-08:
> requests on the go with GitHub Mobile for iOS
> and Android
> Download it today!

Quite possibly, I don’t have any to test

luyii-code-1, 2026-10-08:
https://github.com/luyii-code-1/dji-ocusync-droneid-research/
Here are my stable research results regarding O2. You might have selected the wrong frequency band. When I captured the packets, I set the frequency band in DJI Fly to \"5.8GHz Only,\" and then I was able to capture the DroneID on 2444 and other frequency bands.

King-Of-Knights, 2026-10-08:
> https://github.com/luyii-code-1/dji-ocusync-droneid-research/ Here are my stable research results regarding O2. You might have selected the wrong frequency band. When I captured the packets, I set [...]

5.8GHz only will only influence on video transmission signal, but not droneid

EdwardBlair, 2026-10-08:
Can you upload raw IQ to mega.co.nz or huggingface or file.io?

EdwardBlair, 2026-10-08:
Yes droneid hops on 2.4 and 5.8 regardless I can confirm that individually

luyii-code-1, 2026-10-08:
> 5.8GHz only will only influence on video transmission signal, but not droneid

对的,但是我测出来如果O2机器(老款Mini2)把频段设成5.8G,在2444Mhz抓到DroneID的概率高很多,而且这样能够排除视频信号的干扰

pandajr69, 2026-10-08:
@luyii-code-1 

In the app (DJI Fly), If you set the band 2.4 or 5.8, you set only video and link signal. Drone continue jumping 2.4 and 5.8

mr-aq-stack, 2026-10-08:
Try to capture the downlink signal of the drone. if your Rx side is configured properly and your decoding logic is synced with rx then it decodes successfully.
When you set the freq on controller, you are setting it's uplink. If you off the controller the downlink signal will still receive and it should decode too.
So don't confuse yourself on controller.
And yes downlink signal will continue hopping in both bands but on specific frequency.

[22 more replies]